rhgndf/ms912x #1: Unable to make!!

Dharm94, 2022-10-03:
dharm@dharm-IdeaPad:~/Desktop/ms912x$ sudo ./insmod.sh 
[sudo] password for dharm: 
make -C /lib/modules/5.15.0-48-generic/build M=/home/dharm/Desktop/ms912x clean
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-48-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-48-generic'
rm -f /home/dharm/Desktop/ms912x/Module.symvers /home/dharm/Desktop/ms912x/*.ur-safe
make CHECK="/usr/bin/sparse" -C /lib/modules/5.15.0-48-generic/build M=/home/dharm/Desktop/ms912x modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-48-generic'
  CC [M]  /home/dharm/Desktop/ms912x/ms912x_registers.o
  CC [M]  /home/dharm/Desktop/ms912x/ms912x_connector.o
  CC [M]  /home/dharm/Desktop/ms912x/ms912x_transfer.o
  CC [M]  /home/dharm/Desktop/ms912x/ms912x_drv.o
  LD [M]  /home/dharm/Desktop/ms912x/ms912x.o
  MODPOST /home/dharm/Desktop/ms912x/Module.symvers
  CC [M]  /home/dharm/Desktop/ms912x/ms912x.mod.o
  LD [M]  /home/dharm/Desktop/ms912x/ms912x.ko
  BTF [M] /home/dharm/Desktop/ms912x/ms912x.ko
Skipping BTF generation for /home/dharm/Desktop/ms912x/ms912x.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-48-generic'
rmmod: ERROR: Module ms912x is not currently loaded

rhgndf, 2022-10-03:
The error can be safely ignored.

Dharm94, 2022-10-03:
Can you list your kernal !!!!!!!

Bcoz only once i was able to see flickered screen on my second monitor

Now it's not showing

Btw i'm using dual boot 22.04,

Dharm94, 2022-10-03:
Hey ! When it worked at that time output was like :

[image]

Now when i run make all -j [The last line is not coming and even nothing is working now.]

[image]

rhgndf, 2022-10-03:
I use kernel 5.17. It works on 5.15 for me, not sure what's wrong. Try a reboot?

Dharm94, 2022-10-04:
Worked on kernel 5.15.0.47 & ****.48 !!! But still FPS is very low <,,

But thank you So SO So much ;)

closed 2022-10-04 by Dharm94