rhgndf/ms912x #2: This driver don't work with linux-5.19.0

yoseforb, 2023-06-27:
I would like to know what is your linux-kernel version that work with this driver.

With linux-5.19.0 I changed dma_buf_map to iosys_map to compile the driver:

But I got this error after running ./insmod.sh:
`insmod: ERROR: could not insert module ms912x.ko: Unknown symbol in module
`
from dmseg:
```
[  256.959872] ms912x: loading out-of-tree module taints kernel.
[  256.959910] ms912x: module verification failed: signature and/or required key missing - tainting kernel
[  256.960030] ms912x: Unknown symbol drm_gem_shmem_dumb_create (err -2)
[  256.960250] ms912x: Unknown symbol drm_gem_shmem_prime_import_sg_table (err -2)
```

lenainjaune, 2023-09-03:
@yoseforb 
> With linux-5.19.0 I changed dma_buf_map to iosys_map to compile the driver:

Have you really managed to compile ?

In my case in a fresh Debian 11 with 5.10.0-20-amd64 kernel and its headers (default was 5.10.0-9), with make I get :
```sh
root@vm-bullseye-xfce:~/ms912x-main# make
make CHECK="/usr/bin/sparse" -C /lib/modules/5.10.0-20-amd64/build M=/root/ms912x-main modules
make[1] : on entre dans le répertoire « /usr/src/linux-headers-5.10.0-20-amd64 »
  CC [M]  /root/ms912x-main/ms912x_registers.o
In file included from /root/ms912x-main/ms912x_registers.c:4:
/root/ms912x-main/ms912x.h:113:19: warning: ‘struct dma_buf_map’ declared inside parameter list will not be visible outside of this definition or declaration
  113 |      const struct dma_buf_map *map,
      |                   ^~~~~~~~~~~
  CC [M]  /root/ms912x-main/ms912x_connector.o
In file included from /root/ms912x-main/ms912x_connector.c:7:
/root/ms912x-main/ms912x.h:113:19: warning: ‘struct dma_buf_map’ declared inside parameter list will not be visible outside of this definition or declaration
  113 |      const struct dma_buf_map *map,
      |                   ^~~~~~~~~~~
  CC [M]  /root/ms912x-main/ms912x_transfer.o
In file included from /root/ms912x-main/ms912x_transfer.c:7:
/root/ms912x-main/ms912x.h:113:19: warning: ‘struct dma_buf_map’ declared inside parameter list will not be visible outside of this definition or declaration
  113 |      const struct dma_buf_map *map,
      |                   ^~~~~~~~~~~
/root/ms912x-main/ms912x_transfer.c:160:19: warning: ‘struct dma_buf_map’ declared inside parameter list will not be visible outside of this definition or declaration
  160 |      const struct dma_buf_map *map,
      |                   ^~~~~~~~~~~
/root/ms912x-main/ms912x_transfer.c:159:6: error: conflicting types for ‘ms912x_fb_send_rect’
  159 | void ms912x_fb_send_rect(struct drm_framebuffer *fb,
      |      ^~~~~~~~~~~~~~~~~~~
In file included from /root/ms912x-main/ms912x_transfer.c:7:
/root/ms912x-main/ms912x.h:112:6: note: previous declaration of ‘ms912x_fb_send_rect’ was here
  112 | void ms912x_fb_send_rect(struct drm_framebuffer *fb,
      |      ^~~~~~~~~~~~~~~~~~~
/root/ms912x-main/ms912x_transfer.c: In function ‘ms912x_fb_send_rect’:
/root/ms912x-main/ms912x_transfer.c:164:19: error: invalid use of undefined type ‘const struct dma_buf_map’
  164 |  void *vaddr = map->vaddr;
      |                   ^~
/root/ms912x-main/ms912x_transfer.c:195:8: error: implicit declaration of function ‘drm_gem_fb_begin_cpu_access’; did you mean ‘dma_buf_begin_cpu_access’? [-Werror=implicit-function-declaration]
  195 |  ret = drm_gem_fb_begin_cpu_access(fb, DMA_FROM_DEVICE);
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        dma_buf_begin_cpu_access
/root/ms912x-main/ms912x_transfer.c:237:2: error: implicit declaration of function ‘drm_gem_fb_end_cpu_access’; did you mean ‘dma_buf_end_cpu_access’? [-Werror=implicit-function-declaration]
  237 |  drm_gem_fb_end_cpu_access(fb, DMA_FROM_DEVICE);
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~
      |  dma_buf_end_cpu_access
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-5.10.0-20-common/scripts/Makefile.build:291 : /root/ms912x-main/ms912x_transfer.o] Erreur 1
make[2]: *** [/usr/src/linux-headers-5.10.0-20-common/Makefile:1861 : /root/ms912x-main] Erreur 2
make[1]: *** [/usr/src/linux-headers-5.10.0-20-common/Makefile:185 : __sub-make] Erreur 2
make[1] : on quitte le répertoire « /usr/src/linux-headers-5.10.0-20-amd64 »
make: *** [Makefile:15 : modules] Erreur 2
```
and as far as I know `error: conflicting types` can not be ignored, so I ask to myself how you achieved this , with a different kernel and with a configuration hack.

lenainjaune, 2023-09-05:
Before digging, tweaking and make a better fitted solution I want to announce to you that we have deduced that this version of the module is dedicated for **Ubuntu 20.04 LTS** with the **v5.15.0-43** kernel (the one tested with success for now). We found that in digging some elements that you indicated in your post (**dma_buf__map** to **iosys_map**) which made us in the right direction. I opened a more detailed topic to explain all this [on a french generalist forum](https://forums.commentcamarche.net/forum/affich-37906185-erreur-de-compilation-conflicting-types?GdRIGoe5eugmw9n9GU5NEqjnqZyJf8M6dy0GGx25GI8#). I tested with success to plug another monitor in a virtual machine (Qemu/KVM) : I can configure a mirror with the virtual system screen and an external monitor.

DSergeev17606, 2023-09-06:
Для сборки подходит любое ядро от 5.11 до 5.17.
Но работает плохо. В сессии x11 особенно плохо.
[image]

lenainjaune, 2023-09-24:
> I would like to know what is your linux-kernel version that work with this driver.
>
> With linux-5.19.0 I changed dma_buf_map to iosys_map to compile the driver:
>
> But I got this error after running [...]

Hi ! 

If you try this :

```
root@host:~/ms912x-main# /usr/sbin/modinfo ./ms912x.ko
filename:       /root/ms912x-main/./ms912x.ko
license:        GPL
alias:          usb:v534Dp0821d*dc*dsc*dp*icFFisc00ip00in*
alias:          usb:v534Dp6021d*dc*dsc*dp*icFFisc00ip00in*
depends:        drm_kms_helper,drm,usbcore,drm_shmem_helper
retpoline:      Y
name:           ms912x
vermagic:       6.1.0-10-amd64 SMP preempt mod_unload modversions 
```

In my case the **drm_shmem_helper** dependency is not loaded, as indicated below :

```
root@host:~/ms912x-main# lsmod \
| grep -Eo \
 "^(drm_kms_helper|drm|usbcore|drm_shmem_helper)" \
| sort -u
drm
drm_kms_helper
usbcore
```

Then I load it and now I can load the module ms912x without error :

```
root@host:~/ms912x-main# modprobe drm_shmem_helper

root@host:~/ms912x-main# lsmod \
| grep -c drm_shmem_helper
2

root@host:~/ms912x-main# insmod ./ms912x.ko

root@host:~/ms912x-main# journalctl -k | grep ms912x
sept. 23 00:23:08 host kernel: ms912x: \\
 loading out-of-tree module taints kernel.
sept. 23 00:23:08 host kernel: ms912x: \\
 module verification failed: \\
 signature and/or required key missing - tainting kernel
sept. 23 00:23:08 host kernel: \\
 usbcore: registered new interface driver ms912x
```

See also [the thread I opened](https://forums.commentcamarche.net/forum/newest/37906185-erreur-de-compilation-conflicting-types#p37915795) (in french sorry)

Good luck :-)

With adelphity
lnj

rhgndf, 2024-01-11:
Since the code is updated for newer kernel, closing this issue.

closed 2024-01-11 by rhgndf