rhgndf/ms912x #3: Update sources.

DSergeev17606, 2023-08-22:
Please update the sources to compile with kernel 6+

nunesbns, 2023-10-05:
https://github.com/nunesbns/ms912x

Project with kernel 6 support.

faveoled, 2023-10-10:
@nunesbns Getting this error:
```
/home/user/Downloads/ms912x/ms912x_connector.c:12:30: error: ‘EDID_LENGTH’ undeclared (first use in this function)
   12 |         int offset = block * EDID_LENGTH;
      |                              ^~~~~~~~~~~
```
Any ideas?

nunesbns, 2023-10-11:
@faveoled move to kernel-6-support branch. 

https://github.com/nunesbns/ms912x/tree/kernel-6-support

faveoled, 2023-10-11:
@nunesbns Compiled fine but has this error on insert:
```
insmod: ERROR: could not insert module ms912x.ko: Unknown symbol in module
```
Had this warning in compile:
```
Skipping BTF generation for /home/user/Downloads/ms912x-kernel-6-support/ms912x.ko due to unavailability of vmlinux
```
Any ideas?

lenainjaune, 2023-10-11:
Hi :)

@faveoled

From a fresh **Debian 12 Bookworm** system with the default kernel **v6.1.0-10-amd64** (virtual environment for tests).

> insmod: ERROR: could not insert module ms912x.ko: Unknown symbol in module

In my case, infos gave me :
```sh
root@vm-bookworm:/ms912x-main# /usr/sbin/modinfo ./ms912x.ko
filename:       /root/ms912x-main/./ms912x.ko
license:        GPL
alias:          usb:v534Dp0821d*dc*dsc*dp*icFFisc00ip00in*
alias:          usb:v534Dp6021d*dc*dsc*dp*icFFisc00ip00in*
depends:        drm_kms_helper,drm,usbcore,drm_shmem_helper
retpoline:      Y
name:           ms912x
vermagic:       6.1.0-10-amd64 SMP preempt mod_unload modversions
```
You can see there was 4 dependencies : drm_kms_helper,drm,usbcore,drm_shmem_helper

In my case **drm_shmem_helper** was not loaded by default, as you can see here :
```sh
root@vm-bookworm:/ms912x-main# lsmod \
| grep -Eo \
 "^(drm_kms_helper|drm|usbcore|drm_shmem_helper)" \
| sort -u
drm
drm_kms_helper
usbcore
```

So I tried to load **drm_shmem_helper** :
```sh
root@vm-bookworm:/ms912x-main# modprobe drm_shmem_helper
root@vm-bookworm:/ms912x-main# lsmod | grep -c drm_shmem_helper
2
```
=> The module was loaded.

I finally tried to load the custom module **ms912x** :
```sh
root@vm-bookworm:/ms912x-main# insmod ./ms912x.ko
# Nothing displayed
root@vm-bookworm:/ms912x-main# lsmod | grep -c ms912x
2
root@vm-bookworm:/ms912x-main# journalctl -k | grep ms912x
sept. 23 00:23:08 vm-bookworm kernel: ms912x: \\
 loading out-of-tree module taints kernel.
sept. 23 00:23:08 vm-bookworm kernel: ms912x: \\
 module verification failed: \\
 signature and/or required key missing - tainting kernel
sept. 23 00:23:08 vm-bookworm kernel: \\
 usbcore: registered new interface driver ms912x
```

Success :D !

Note : the **tainting kernel** message is not blocking, you can ignore it ; it could be resolved in signing this custom module (that is apparently the why of this message) ; I searched a way to achieve this but I did terminate as I had more urgent stuff to treat.

> Skipping BTF generation for /home/user/Downloads/ms912x-kernel-6-support/ms912x.ko due to unavailability of vmlinux

As I understood you can ignore it too however you can get rid of this in compiling the module **resolve_btfids** (see [here](https://forums.commentcamarche.net/forum/newest/37906185-erreur-de-compilation-conflicting-types#p37910960) - sorry in french, you must search "Le problème de vmlinux manquant" which will be translated in "The problem of missing vmlinux")

As I said I do not manage this project for the moment as I have some urgent stuff to accomplish before, but you can read what I tried [in this thread](https://forums.commentcamarche.net/forum/newest/37906185-erreur-de-compilation-conflicting-types) (in french but you can use some online translators) with the help of more expert people (this is the first time I dig as much for compiling a project).

Too if you follow the link you could see I modified the project's headers to fit with the kernel v6+ to correct among others the EDID's problem you notified.

I hope this will help you ;)

skylight74, 2023-10-12:
first of all thanks for the update @nunesbns , and also thanks for @lenainjaune the detailed walkthrough. I have facing an issue on the display it self.
[image]

My system details are: 
Linux-kernel-6.5.6.1-cachyos-cfs (Arch linux based distro)
Intel Corporation Skylake GT2 [HD Graphics 520] 
NVIDIA Corporation GM108M [GeForce 940MX]
Hyprland wm with a wayland compositor.
2 external monitors, HDMI+ USB/VGA , and one internal Monitor.

I'm relatively new to linux driver development, so debugging the issue myself proved challenging. 

So far its loading properly, and even works with Plymouth loads, but that's it.

If you could assist me fix this problem, whether if it is a compositor issue (wayland/Hyprland) ,driver issue/ or GPU issue . I would be very grateful. 

and again thanks for everything.

P.s if you need any logs and problem description. please let me know :smile:

lenainjaune, 2023-10-12:
@faveoled 
> @lenainjaune Thanks. It seems to be loading now:

Cool :sunglasses: 

And is the display detected by your system ?

In my case it is detected (even in virtual environment) only if I use Wayland with only a Desktop Environment which supports it natively (Gnome, KDE, Cinnamon but not yet for XFCE) both with a Display Manager which support it (GDM3, SDDM but not completely with LightDM).

As I use intensively XFCE4 with LighDM (my prefered DE/DM environment), I am stuck for now. But I do not despere to find a solution or a workaround ... when I will dig more deeper.

@skylight74 
> first of all thanks for the update @nunesbns , and also thanks for @lenainjaune the detailed walkthrough. I have facing an issue on the display it self.

Research and findings are meant to be shared, it seems to me. In any case it is a value that speaks to me :wink: 

> I have facing an issue on the display it self.

Nice :laughing: are you sure you are on Arch as it feels to me distinguish a Debian logo part :laughing: 

More seriously I do not know how to resolve or debug your problem. My feeling as you suggest is to orientate your investigation's domain on the video card drivers. Do you use the nVidia [nouveau](https://nouveau.freedesktop.org/) driver or the proprietary one ? Could you try both and see if there are differences ?

In all case I will let people with more experience on this subject speak.

skylight74, 2023-10-15:
@lenainjaune thank you very much, I'm using the proprietary one, and the logo is the plymouth theme. the one that i mentioned were working earlier. I don't really have a Display manager. just A window manager as I usually create my sessions from the terminal (de bloating the distro.

I will debug by offloading everything to the main GPU :crossed_fingers:  it might be helpful

and thank you for your reply :smile:

rhgndf, 2023-10-15:
The code now compiles with kernel 6.5.5. 

@nunesbns Mind if you PR your changes?

YSpoof, 2023-10-16:
> The code now compiles with kernel 6.5.5.

```
  CC [M]  /dev/shm/ms912x/ms912x_drv.o
/dev/shm/ms912x/ms912x_drv.c:13:10: fatal error: drm/drm_fbdev_generic.h: No such file or directory
   13 | #include <drm/drm_fbdev_generic.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

Debian 12
Kernel: 6.1.0-13-amd64

Any tips ?

rhgndf, 2023-10-16:
Try removing that line.

YSpoof, 2023-10-16:
> Try removing that line.

`insmod: ERROR: could not insert module ms912x.ko: Unknown symbol in module`

```
/dev/shm/ms912x main*
❯ file ms912x.ko 
ms912x.ko: ELF 64-bit LSB relocatable, x86-64, version 1 (SYSV), BuildID[sha1]=c9934fabc385055d02d6bb46330b04e075497250, with debug_info, not stripped
```

Btw, total noob here in terms of kernel/module stuff :(

rhgndf, 2023-10-16:
Try: modprobe drm_shmem_helper

YSpoof, 2023-10-16:
> Try: modprobe drm_shmem_helper

Thanks, worked but now a few strange things happened

1. System became very slow with the adapter plugged in
2. I can see my desktop on the second monitor but it is flicking a lot and also shaking the image, especially when updates happens to screen ( like mouse movement )

Similar problem as @skylight74 , but as I said, I "can see" my desktop

Yes, I tried playing with the resolutions, but same deal...

Also, my bad, I should've read @lenainjaune comment

rhgndf, 2023-10-17:
1. Unfortunately, I could not reproduce the issue on my side, so if you or someone who experiences this issue could debug it, that would be helpful.
2. I have seen this happening, but not as severe as you describe it. I think for my side it could be due to the desktop environment: Gnome/KDE/etc... Not totally sure whats causing it in general though, so debugging help is welcome.

rhgndf, 2023-10-20:
For 2 it might be fixed with this commit: ac7e2a5d38c765de21e8df646c969a03c6175422

rhgndf, 2023-10-23:
For 1, it might be because of latency. Might be fixed with b763760e9c83bcc299ebd08972d79e7bebbc3c2d

rhgndf, 2024-01-11:
Since the code is updated for newer kernel, closing this issue.

closed 2024-01-11 by rhgndf