rhgndf/ms912x #4: Image distortion at native display resolution

aivx, 2023-10-15:
The driver works well on Nobara Linux 38 6.5.6-200.fsync.fc 38.x86_64
but there is a distortion on the native display resolution, other resolutions work fine
[image]
[image]
[image]

rhgndf, 2023-10-16:
I don't have the setup to test 1360x768, so not sure what's going on. It could be because its forcing a 1366x768 mode instead.

aivx, 2023-10-18:
yes, I have only 1366 in the list of permissions,
I tried to add an entry from 1360 to the ms912x_mode_list[] list from the resolutions file, but it still does not appear in the system, only 1366

rhgndf, 2023-10-20:
It might be fixed with 4f8fab3d4704c82f902607dfc500737bf6ed7b02.

aivx, 2023-10-20:
it works, but there is a small stripe on the right side
[image]

rhgndf, 2023-10-21:
The hardware seems to be only able to handle multiples of 16. Without more documentation I not sure how to fix this.

rhgndf, 2026-09-05:
fixed in the latest update

closed 2026-09-05 by rhgndf