rhgndf/ms912x #5: Fatal error: Invalid --compress-debug-sections option: `zstd'

eduardolucioac, 2023-10-17:
Any ideas on how to resolve this issue?

I'm using Manjaro with Linux Kernel 6.5.5-1.

Thanks! S2

```
[eduardolac@eduardolac-pc ms912x]$ make all
make CHECK="/usr/bin/sparse" -C /lib/modules/6.5.5-1-MANJARO/build M=/home/eduardolac/Data1/Temp/20231016.1224.0/ms912x modules
make[1]: Entering directory '/usr/lib/modules/6.5.5-1-MANJARO/build'
  CC [M]  /home/eduardolac/Data1/Temp/20231016.1224.0/ms912x/ms912x_registers.o
Assembler messages:
Fatal error: Invalid --compress-debug-sections option: `zstd'
make[3]: *** [scripts/Makefile.build:243: /home/eduardolac/Data1/Temp/20231016.1224.0/ms912x/ms912x_registers.o] Error 1
make[2]: *** [/usr/lib/modules/6.5.5-1-MANJARO/build/Makefile:2034: /home/eduardolac/Data1/Temp/20231016.1224.0/ms912x] Error 2
make[1]: *** [Makefile:234: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.5.5-1-MANJARO/build'
make: *** [Makefile:15: modules] Error 2
```

rhgndf, 2023-10-17:
Unfortunately I have no idea, since I tested it on manjaro 6.5.5 also. Maybe try installing zstd packages?

eduardolucioac, 2023-10-17:
My "ld"

Does'nt support

eduardolucioac, 2023-10-17:
My "ld" doesn't support "zstd"... See..

```
[eduardolac@eduardolac-pc ms912x]$ ld --help | grep "compress-debug sections"
    --compress-debug-sections=[none|zlib|zlib-gnu|zlib-gabi]
```

But, if I use Kernel 6.6 ("linux-amd-drm-fixes" and "linux-amd-drm-fixes-headers" packages in Manjaro), I can build "ms912x".

But, Kernel 6.6 is "experimental"...

eduardolucioac, 2023-10-18:
Solution here: https://forum.manjaro.org/t/binutils-kernel-6-5-x-invalid-compress-debug-sections-option-zstd/149943/8 . 😘

closed 2023-10-18 by eduardolucioac