rhgndf/ms912x #6: The device is recognized, but the screen is completely dark 😢

eduardolucioac, 2023-10-18:
The device is recognized, but the screen is completely dark.

If I restart OS, then the driver no longer works. I have to reinstall.

Any ideas what I can do? 😢

Manjaro Linux (Linux Kernel 6.5.5)

rhgndf, 2023-10-19:
Can you change the resolutions in the display settings? What resolutions are available? Maybe try changing them.

eduardolucioac, 2023-10-19:
*Can you change the resolutions in the display settings?* -> Yes, but the monitor does not work with any.

*What resolutions are available?* -> See the image.

*Maybe try changing them.* -> I already did, but it didn't work.

**NOTE:** When I try to activate the "ms912x" monitor, the other monitors are unconfigured for a moment and then the "ms912x" monitor is deactivated.

**PLUS I:** This "fork" contains implementations that are not available in its repository ( https://github.com/nunesbns/ms912x/tree/kernel-6-support ). I think it would be in your interest to take a look.

Is it your knowledge whether the "ms912x" driver works on Manjaro Linux?

**PLUS II:** I have the original drivers for Android (ARM) if you are interested.

**PLUS III:** Here ( https://askubuntu.com/a/1488791/134723 ) there is an interesting forum about the "ms912x" driver.

Thanks! 🤗

rhgndf, 2023-10-20:
I have fixed a few bugs, try with the new version of the driver?

rhgndf, 2023-10-20:
I tested on more monitors and different setup and couldn't replicate. I also have no idea how to start since I don't have any guesses what might cause it.

eduardolucioac, 2023-10-20:
I reproduce the process on a clean Manjaro KDE (23.0.4, linux65) installation. Same problem...

If we had some log output to be able to handle this problem... Then we would have some starting point to find out what happens... 🙂

Just to clarify further, this device uses the Macrosilicon MS9122 chipset ( http://en.macrosilicon.com/info.asp?base_id=2&third_id=59 , https://blog.csdn.net/YOU_SeMi/article/details/125105651 ).

Here's a photo and the purchase link...

[image]

https://pt.aliexpress.com/item/1005005472898898.html?spm=a2g0o.order_list.order_list_main.5.2bcacaa4USGRdN&gatewayAdapt=glo2bra

eduardolucioac, 2023-10-20:
```
[eduardolac@eduardolac-pc ms912x]$ journalctl -k | grep ms912x
out 20 16:35:49 eduardolac-pc kernel: usbcore: registered new interface driver ms912x
out 20 16:37:07 eduardolac-pc kernel: [drm] Initialized ms912x 0.0.1 20220101 for 1-4:1.3 on minor 2
out 20 16:37:07 eduardolac-pc kernel: ms912x 1-4:1.3: [drm] Cannot find any crtc or sizes
out 20 16:37:07 eduardolac-pc kernel: ms912x 1-4:1.3: [drm] Cannot find any crtc or sizes
out 20 16:37:08 eduardolac-pc kernel: ms912x 1-4:1.3: [drm] fb1: ms912xdrmfb frame buffer device
```

My last "ms912x" driver installation did not cause "video crash", but the monitor screen remained black.

eduardolucioac, 2023-10-20:
I sent this email to MacroSilicon. If it doesn't work, I at least tried...

"""
support@macrosilicon.com, suggestion@macrosilicon.com

An open source driver is being built for devices based on MS912X family chips

Greetings!

    An open source driver ( https://github.com/rhgndf/ms912x ) (Linux Systems) is being built for devices based on MS912X family chips ( http://en.macrosilicon.com/info.asp?base_id=2&third_id=59 ).

    I ask if any support from you is possible or even if there is the possibility of making the Source Codes of the Android drivers ("MSDisplay_Android") available to us so that we can compile for the "AMD64" architecture ("x86_64")

We would appreciate any help! 🤗
"""

rhgndf, 2023-10-21:
Just to check this monitor works if you use your laptop/desktop hdmi port? Check for cabling issues also? 

Could you try attach the lsusb -v output for your dongle? This is to check for differences.

eduardolucioac, 2023-10-22:
_Just to check this monitor works if you use your laptop/desktop hdmi port?_ -> The monitor is working, I've already done the test.

_Check for cabling issues also?_ -> I tested it on Windows and it worked correctly.

The `lsusb -v` output...

```
Bus 001 Device 006: ID 534d:6021 MacroSilicon VGA Display Adapter
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x534d MacroSilicon
  idProduct          0x6021 VGA Display Adapter
  bcdDevice            1.10
  iManufacturer           1 USB Display 
  iProduct                2 usb extscreen
  iSerial                 3 2019BA7160B0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00a2
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode           33 US
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      23
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              16
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         1
      bInterfaceCount         2
      bFunctionClass          1 Audio
[... 107 lines cut ...]
          bDescriptorSubtype      1 (EP_GENERAL)
          bmAttributes         0x00
          bLockDelayUnits         0 Undefined
          wLockDelay         0x0000
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              5 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
```

rhgndf, 2023-10-23:
I don't see a difference between the one I have and yours, so out of ideas here.

eduardolucioac, 2023-10-23:
When you used Linux Manjaro (Linux 6.5.X) to test, did you use the KDE version? For some reason I think this may have an influence... Could you do this test? 🤗

rhgndf, 2023-10-24:
I tried with KDE. With X11 it didn't work, but with wayland the display is detected. Not sure why also.

eduardolucioac, 2023-10-24:
Some information that could be useful to understand the problem...

 - `dmesg | grep -Ei "ms912x"`

```
[62239.612504] usbcore: registered new interface driver ms912x
[62305.617411] [drm] Initialized ms912x 0.0.1 20220101 for 1-4:1.3 on minor 2
[62305.617597] ms912x 1-4:1.3: [drm] Cannot find any crtc or sizes
[62305.618091] ms912x 1-4:1.3: [drm] Cannot find any crtc or sizes
[62325.729553] ms912x 1-4:1.3: [drm] fb1: ms912xdrmfb frame buffer device
```

 - `journalctl -k | grep -Ei "ms912x"`

```
out 24 08:49:42 eduardolac-pc kernel: usbcore: registered new interface driver ms912x
out 24 08:50:48 eduardolac-pc kernel: [drm] Initialized ms912x 0.0.1 20220101 for 1-4:1.3 on minor 2
out 24 08:50:48 eduardolac-pc kernel: ms912x 1-4:1.3: [drm] Cannot find any crtc or sizes
out 24 08:50:48 eduardolac-pc kernel: ms912x 1-4:1.3: [drm] Cannot find any crtc or sizes
out 24 08:51:08 eduardolac-pc kernel: ms912x 1-4:1.3: [drm] fb1: ms912xdrmfb frame buffer device
```

 - `edid-decode < /sys/class/drm/card2-HDMI-A-2/edid`

```
edid-decode (hex):

00 ff ff ff ff ff ff 00 05 e3 01 22 ba 03 00 00
1c 21 01 03 80 30 1a 78 2a 55 d5 ac 50 49 a3 26
13 50 54 bf ef 00 d1 c0 b3 00 95 00 81 80 81 40
81 c0 01 01 01 01 02 3a 80 18 71 38 2d 40 58 2c
45 00 df 04 11 00 00 1e 00 00 00 ff 00 57 58 44
4e 37 58 41 30 30 30 39 35 34 00 00 00 fc 00 32
32 42 31 57 47 35 0a 20 20 20 20 20 00 00 00 fd
00 30 4b 1e 55 12 00 0a 20 20 20 20 20 20 01 f5

02 03 29 f1 4b 10 1f 05 14 04 13 03 12 02 11 01
23 09 07 07 83 01 00 00 67 03 0c 00 10 00 00 3c
68 1a 00 00 01 01 30 4b e6 2a 44 80 a0 70 38 27
40 30 20 35 00 df 04 11 00 00 1a 01 1d 00 72 51
d0 1e 20 6e 28 55 00 df 04 11 00 00 1e 8c 0a d0
8a 20 e0 2d 10 10 3e 96 00 df 04 11 00 00 18 8c
0a d0 90 20 40 31 20 0c 40 55 00 df 04 11 00 00
18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 1c

----------------

Block 0, Base EDID:
  EDID Structure Version & Revision: 1.3
  Vendor & Product Identification:
    Manufacturer: AOC
    Model: 8705
    Serial Number: 954
    Made in: week 28 of 2023
  Basic Display Parameters & Features:
    Digital display
    Maximum image size: 48 cm x 26 cm
    Gamma: 2.20
    DPMS levels: Off
    RGB color display
    First detailed timing is the preferred timing
  Color Characteristics:
    Red  : 0.6728, 0.3134
    Green: 0.2861, 0.6376
    Blue : 0.1513, 0.0751
    White: 0.3134, 0.3291
  Established Timings I & II:
    IBM     :   720x400    70.081663 Hz   9:5     31.467 kHz     28.320000 MHz
    DMT 0x04:   640x480    59.940476 Hz   4:3     31.469 kHz     25.175000 MHz
    Apple   :   640x480    66.666667 Hz   4:3     35.000 kHz     30.240000 MHz
[... 88 lines cut ...]
    Option "AllowNVIDIAGPUScreens"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    BusID "PCI:0:2:0"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    BusID "PCI:1:0:0"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
EndSection
```

rhgndf, 2023-10-24:
Can you try the latest commit? Fixed some latency issue in b763760e9c83bcc299ebd08972d79e7bebbc3c2d. Also since this is a USB2 device, the bandwidth is limited if there are too many updates to be drawn on the screen.

rhgndf, 2023-10-24:
Did you add in the VIDs and PIDs in #7 in? So far I don't have any USB 3.0 devices to test, only USB 2.0 devices with just a blue connector.

rhgndf, 2023-10-24:
The better performances part is interesting, without more documentation, or capturing the wireshark usb traces I currently don't have a clue whats causing it.

eduardolucioac, 2023-10-24:
_Did you add in the VIDs and PIDs in #7 in?_ -> I tested these modifications, but nothing changed.
_capturing the wireshark usb traces I currently don't have a clue whats causing it._ -> I don't know how to provide you with this information.

rhgndf, 2023-10-25:
I guess there's not much I can do for now, sorry about that. In the meantime can you check whether it's an actual USB 3 device? Previously you mentioned it's a MS9122 but that's only a USB 2 device.

eduardolucioac, 2023-10-26:
_MS9122 but that's only a USB 2 device_ -> Yes! It is a USB 2.0 device (fake USB 3.0).

Thank you for your help!
At least we tried... 😊

Long live Free Software! ❤️🐧

rhgndf, 2023-11-02:
@eduardolucioac Could you try the new improvements? I might have partially fixed it, if not fully.

eduardolucioac, 2023-11-03:
If it's not too much to ask... Is it now working with KDE using X11? Wayland is simply unusable... 😅

rhgndf, 2023-11-03:
I'm not sure why it doesn't work. Could be a KDE/X11 issue? I have experienced a KDE crash when unplugging the usb. Tried comparing it with other drivers but still not sure why it doesn't work. I'm sorry I don't have any more insight into this.

eduardolucioac, 2023-11-03:
_I have experienced a KDE crash when unplugging the usb._ -> Same, here!

rhgndf, 2023-11-03:
Then it is likely to be a KDE/X11 issue.

closed 2023-11-03 by rhgndf

eduardolucioac, 2023-11-03:
Work, as spected. But, with a very low performance, again...
Again, Wayland has a lot of issues.
Again, on Windows we had exponentially better performances.

Thanks! =]

rhgndf, 2023-11-03:
The slowness has probably to do with reading from the framebuffer which is write combining, which means reads are not cached and slow. I used a memcpy to try to speed it up but it's still slow. Not sure how to proceed from here.

For the KDE/X11 you might want to file a bug report on that side.

s12wu, 2023-11-03:
Same crash here under X11 with XFCE (Manjaro ARM), so it probably isn't related to KDE

Xorg.0.log
```
[  1053.458] (II) config/udev: removing GPU device /sys/devices/platform/soc/ffe09000.usb/ff500000.usb/xhci-hcd.0.auto/usb1/1-1/1-1.3/1-1.3:1.3/drm/card2 /dev/dri/card2
[  1053.466] xf86: remove device 2 /sys/devices/platform/soc/ffe09000.usb/ff500000.usb/xhci-hcd.0.auto/usb1/1-1/1-1.3/1-1.3:1.3/drm/card2
[  1053.467] (EE) 
[  1053.467] (EE) Backtrace:
[  1053.483] (EE) 0: /usr/lib/Xorg (OsLookupColor+0x188) [0xaaaac7d074a8]
[  1053.528] (EE) unw_get_proc_info failed: no unwind info found [-10]
[  1053.528] (EE) 
[  1053.529] (EE) Segmentation fault at address 0x0
[  1053.532] (EE) 
Fatal server error:
[  1053.532] (EE) Caught signal 11 (Segmentation fault). Server aborting
```

rhgndf, 2023-11-03:
From the log it looks like X11 issue.

rapgro, 2025-09-16:
Well, we'd need some udev rule(s). What are the device ids in support? As I own an USB/DVI adapter, I can test with that.