rhgndf/ms912x #7: request ms9132: quite successful already with just adding missing MODULE_ALIAS and USB_DEVICE_AND_INTERFACE_INFO

jakobi, 2023-10-19:
Request to add ms9132.

Both request and a heads  up  for a simple USB3-A to HDMI dongle (based  
on  Macrosilicon ms9132, usb  345f:9132)  showing only some rectangular  artefacts
(speed permitting; possibly the artifacts are kind of compression restarts?).

### Procedure

1. make it compile for the kernel in use: For 6.1 e.g. select a late 2022 commit,
    suitably switch between dma_buf_map and iosys_map, add missing
    include  drm/drm_edid.h) 
    _(this is for a Kali 2023.1 stick, stuck at kernel 6.1, as I **cannot** update the kernel),_  
3. modprobe drm_shmem_helper, 
4. add v345Fp9132 in MODULE_ALIAS and id_table
5. run insmod.sh to recompile and test in X11

### Relevant changes

```c
diff 'dd--color=auto' -u -r ms912x/ms912x.mod.c ms912xMOD/ms912x.mod.c
--- ms912x/ms912x.mod.c 2023-10-18 20:21:16.700507428 +0000
+++ ms912xMOD/ms912x.mod.c 2023-10-19 01:19:16.422745923 +0000
@@ -116,3 +116,4 @@
 
 MODULE_ALIAS("usb:v534Dp6021d*dc*dsc*dp*icFFisc00ip00in*");
 MODULE_ALIAS("usb:v534Dp0821d*dc*dsc*dp*icFFisc00ip00in*");
+MODULE_ALIAS("usb:v345Fp9132d*dc*dsc*dp*icFFisc00ip00in*");
diff '--color=auto' -u -r ms912x/ms912x_drv.c ms912xMOD/ms912x_drv.c
--- ms912x/ms912x_drv.c 2023-10-18 20:04:13.723435640 +0000
+++ ms912xMOD/ms912x_drv.c 2023-10-19 00:29:41.244899105 +0000
@@ -273,6 +273,7 @@
 static const struct usb_device_id id_table[] = {
   { USB_DEVICE_AND_INTERFACE_INFO(0x534d, 0x6021, 0xff, 0x00, 0x00) },
   { USB_DEVICE_AND_INTERFACE_INFO(0x534d, 0x0821, 0xff, 0x00, 0x00) },
+  { USB_DEVICE_AND_INTERFACE_INFO(0x345f, 0x9132, 0xff, 0x00, 0x00) },
   {},
 };
 MODULE_DEVICE_TABLE(usb, id_table);
```

### Tests in X11

Lenovo T61: Due to usb2 speed, on T61 I see a ton  of
rectangular artefacts, and a stable 1920x1080 image is only shown when
idle.  Usable  for manpages and vi, not for browsing less  than  fully
static pages. Youtube is good for jokes only...

Yoga 720 usb3: I've only rarely artefacts. I seem to be
able  to  trigger them somewhat when moving windows, terminal  use  is
near   perfect,  and  even  youtube  in  the  browser  is   sluggisch,
delayed(1+s), but with near always sane display. Audio-over-hdmi seems
to be missing.

rhgndf, 2023-10-19:
Can you try the branch kernel-6.1.

Nice work on the usb3 side, could you create a pull request for those change?

rhgndf, 2023-10-20:
It might be fixed with ac7e2a5d38c765de21e8df646c969a03c6175422.

rhgndf, 2023-10-23:
Latency issue might be fixed with b763760e9c83bcc299ebd08972d79e7bebbc3c2d

buddyjojo, 2024-03-12:
I tested my ms9132 after adding the vid/pid, audio worked fine and didn't notice too much latency or graphical glitches. Though I did get some screen tearing but I did all my testing on Wayland so could be an issue with that.

rhgndf, 2024-03-12:
Since I don't have a USB 3 device and it looks like this works as is without modification, the vid/pid is added in 738aef1b659927b21be62a71fbe6f296bea86ae0.

closed 2024-03-12 by rhgndf