rhgndf/ms912x #8: Is linux 5.10 support possible?

thenightterorx, 2023-11-10:
I want to see how this device would fare on a sbc called the "milk-v duo," but that board currently only supports linux 5.10.4. 
I know very little about c programming, so even just a pointer (no pun intended) would help a lot.

`lee@archlinux ~/ms912x (main)> make -C /lib/modules/5.10.194-1-lts510/build/ M=$PWD
  CC [M]  /home/lee/ms912x/ms912x_transfer.o
In file included from /home/lee/ms912x/ms912x_transfer.c:7:
/home/lee/ms912x/ms912x.h:113:47: warning: ‘struct dma_buf_map’ declared inside parameter list will not be visible outside of this definition or declaration
  113 |                                  const struct dma_buf_map *map,
      |                                               ^~~~~~~~~~~
/home/lee/ms912x/ms912x_transfer.c:160:47: warning: ‘struct dma_buf_map’ declared inside parameter list will not be visible outside of this definition or declaration
  160 |                                  const struct dma_buf_map *map,
      |                                               ^~~~~~~~~~~
/home/lee/ms912x/ms912x_transfer.c:159:6: error: conflicting types for ‘ms912x_fb_send_rect’; have ‘void(struct drm_framebuffer *, const struct dma_buf_map *, struct drm_rect *)’
  159 | void ms912x_fb_send_rect(struct drm_framebuffer *fb,
      |      ^~~~~~~~~~~~~~~~~~~
/home/lee/ms912x/ms912x.h:112:6: note: previous declaration of ‘ms912x_fb_send_rect’ with type ‘void(struct drm_framebuffer *, const struct dma_buf_map *, struct drm_rect *)’
  112 | void ms912x_fb_send_rect(struct drm_framebuffer *fb,
      |      ^~~~~~~~~~~~~~~~~~~
/home/lee/ms912x/ms912x_transfer.c: In function ‘ms912x_fb_send_rect’:
/home/lee/ms912x/ms912x_transfer.c:164:26: error: invalid use of undefined type ‘const struct dma_buf_map’
  164 |         void *vaddr = map->vaddr;
      |                          ^~
/home/lee/ms912x/ms912x_transfer.c:195:15: error: implicit declaration of function ‘drm_gem_fb_begin_cpu_access’; did you mean ‘dma_buf_begin_cpu_access’? [-Werror=implicit-function-declaration]
  195 |         ret = drm_gem_fb_begin_cpu_access(fb, DMA_FROM_DEVICE);
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |               dma_buf_begin_cpu_access
/home/lee/ms912x/ms912x_transfer.c:237:9: error: implicit declaration of function ‘drm_gem_fb_end_cpu_access’; did you mean ‘dma_buf_end_cpu_access’? [-Werror=implicit-function-declaration]
  237 |         drm_gem_fb_end_cpu_access(fb, DMA_FROM_DEVICE);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
      |         dma_buf_end_cpu_access
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:286: /home/lee/ms912x/ms912x_transfer.o] Error 1
make: *** [Makefile:1832: /home/lee/ms912x] Error 2`

rhgndf, 2023-11-10:
Branch kernel-5.10 might work. I had crashes with the branch, but this might be still useful for you.

thenightterorx, 2023-12-07:
I am getting this when trying to build the latest: 
 CC [M]  /home/lee/modules/5.10/ms912x-kernel-5.10/ms912x_registers.o
  CC [M]  /home/lee/modules/5.10/ms912x-kernel-5.10/ms912x_connector.o
  CC [M]  /home/lee/modules/5.10/ms912x-kernel-5.10/ms912x_transfer.o
/home/lee/modules/5.10/ms912x-kernel-5.10/ms912x_transfer.c: In function ‘ms912x_fb_send_rect’:
/home/lee/modules/5.10/ms912x-kernel-5.10/ms912x_transfer.c:154:23: error: ‘map’ undeclared (first use in this function); did you mean ‘vmap’?
  154 |         void *vaddr = map->vaddr;
      |                       ^~~
      |                       vmap
/home/lee/modules/5.10/ms912x-kernel-5.10/ms912x_transfer.c:154:23: note: each undeclared identifier is reported only once for each function it appears in
make[1]: *** [scripts/Makefile.build:286: /home/lee/modules/5.10/ms912x-kernel-5.10/ms912x_transfer.o] Error 1
make: *** [Makefile:1832: /home/lee/modules/5.10/ms912x-kernel-5.10] Error 2