rhgndf/ms912x #10: Other displays are also slowed down whenever there are changes on the screen connected to the ms912x

thenightterorx, 2023-12-02:
As, I am typing this, I have a terminal emulator on a display connected to an ms912x displaying the output of make (so there is lots of text flying around). On this other display, which is connected to a separate graphics card, everything happens very slowly during periods of a lot of output. Is this to be expected? Or is it just because of how xorg works.

rhgndf, 2023-12-02:
Did you get a USB 2 or USB 3 device? That determines how fast it can update.

rhgndf, 2024-01-05:
I think this might be how Xorg works, I have seen this happen on Xorg but not Wayland.

yoseforb, 2024-01-16:
I tested it on Wayland, it just a bit less slow, but still not good for daily use.

rhgndf, 2024-01-17:
Could you check whether the device is a USB 2 device? If it is there isn't enough bandwidth to transfer the frames fast enough.

thenightterorx, 2024-01-19:
Which wayland wm did you try? I tried sway, hyprland, then gnome+mutter, and only gnome worked. Also there isn't the slowdown referenced in this issue, which is great!

rhgndf, 2024-01-19:
I tried on gnome. Have you confirmed whether sway or hyprland supports hotplugging drm devices? That is one of the main reasons why sometimes the drivers don't work.

thenightterorx, 2024-01-19:
> I tried on gnome. Have you confirmed whether sway or hyprland supports hotplugging drm devices? That is one of the main reasons why sometimes the drivers don't work.

After messing around with it, I ended up doing 3 things, some of which might not really be necessary. The --unsupported-gpu flag, export WLR_DRM_NO_MODIFIERS=1, and export WLR_DRM_DEVICES=/dev/dri/card0:/dev/dri/card1.

closed 2024-01-19 by thenightterorx