rhgndf/ms912x #9: This does not appear in sway

thenightterorx, 2023-12-02:
My adapter shows up in xorg (granted, I've only tried i3) and even in other wayland implementations like gnome, but there is nothing about it when I do swaymsg -t get_outputs. I'm probably just missing something here though so any guidance is appreciated.

rhgndf, 2023-12-02:
First thing to check is if it supports DRM hotplug? Some environments do not have good support for display devices coming up and being removed while it is running.

thenightterorx, 2024-01-19:
After messing around with it, I ended up doing 3 things, some of which might not really be necessary. The --unsupported-gpu flag, export WLR_DRM_NO_MODIFIERS=1, and export WLR_DRM_DEVICES=/dev/dri/card0:/dev/dri/card1.

closed 2024-01-19 by thenightterorx