rhgndf/ms912x #11: How to get non-standard resolutions

tablesturn, 2024-01-06:
I am using the code from here: https://github.com/s12wu/ms912x-python/tree/main

It is based on your driver and works just fine for my purpose (auxiliary display output from Python/Touchdesigner) with standard resolutions (e.g. 1920, 1080, 60, 0x8100, MS912X_PIXFMT_UYVY).
Since there is more activity right here and the problem is not related to the actual implementation, I am posting the problem in this repo.

The problem is that I want to use a display with a non-standard resolution of 400x1280.
It advertises the following EDID information, which is not a CEA mode:
```
DTD 1:   400x1280   59.988901 Hz   5:16    81.585 kHz     43.240000 MHz (10 mm x 192 mm)
Hfront   50 Hsync  30 Hback   50 Hpol N
Vfront   30 Vsync  20 Vback   30 Vpol N
```
Is there a way to get this resolution working, or are the timings hard-coded into the MS912x?

When using the resolution of 400x1280 in combination with any of the mode values from the ms912x_mode_list in ms912x_drv.c, the display just shows gibberish (as it does with Macrosilicon's suspicious Windows driver).

rhgndf, 2024-01-07:
The timings seem to be controlled by the mode number, which is 0x8100 in the example above. I have been getting them from the wireshark usb captures from the windows driver. The CEA numbers seem to work, but I'm not able to guarantee there's no issues since I didn't see them in the wireshark captures. 

There might be other mode numbers from 0-0xFF that you can experiment with. Either that or there might be some additional registers to set to allow custom resolutions. I'm not sure since I don't have the register map.

As a quick test you can get the mode that has the closest clock rate to your mode and replace the width and height with your custom resolution.

tablesturn, 2024-01-07:
Okay, but how do the CEA VIC numbers translate to the mode number?
0x81 would be 129 in decimal, but the CEA table says that the numbers 128-192 are reserved and used in SVD to indicate native timing for numbers 1-64:

https://en.wikipedia.org/wiki/Extended_Display_Identification_Data#CTA_Data_Blocks

Probably I am using the wrong table? Table 3 in the CTA‐861-G standard states the same:

https://web.archive.org/web/20171201033424/https://standards.cta.tech/kwspub/published_docs/CTA-861-G_FINAL_revised_2017.pdf

rhgndf, 2024-01-07:
I don't they they are using the CEA VIC numbers exactly, maybe only the lower numbers. The higher mode numbers doesn't seem to correspond to anything as far as I can tell. I haven't really tried it out but you might be able to quickly try it using the python script in that repository.

Bastindo, 2024-01-08:
Hi,
just wanted to add that I found this list of mode numbers when decompiling the ms9120 android driver apk with apktool (in `smali/com/ms/ms2160/CSharpTool/Util$_E_AS7160_VIDEO_FORMAT_.smali`). Havent tested any of them but they seem to line up with the numbers in this driver.

```
VFMT_CEA_NULL 				0x0
VFMT_CEA_01_640x480P_60HZ 		0x1
VFMT_CEA_02_720x480P_60HZ 		0x2
VFMT_CEA_03_720x480P_60HZ 		0x3
VFMT_CEA_04_1280x720P_60HZ 		0x4
VFMT_CEA_05_1920x1080I_60HZ	 	0x5
VFMT_CEA_06_720x480I_60HZ 		0x6
VFMT_CEA_07_720x480I_60HZ 		0x7
VFMT_CEA_08_720x240P_60HZ 		0x8
VFMT_CEA_09_720x240P_60HZ 		0x9
VFMT_CEA_10_720x480I_60HZ 		0xa
VFMT_CEA_11_720x480I_60HZ 		0xb
VFMT_CEA_12_720x240P_60HZ 		0xc
VFMT_CEA_13_720x240P_60HZ 		0xd
VFMT_CEA_14_1440x480P_60HZ 		0xe
VFMT_CEA_15_1440x480P_60HZ 		0xf
VFMT_CEA_16_1920x1080P_60HZ 		0x10
VFMT_CEA_17_720x576P_50HZ 		0x11
VFMT_CEA_18_720x576P_50HZ 		0x12
VFMT_CEA_19_1280x720P_50HZ 		0x13
VFMT_CEA_20_1920x1080I_50HZ 		0x14
VFMT_CEA_21_720x576I_50HZ	 	0x15
VFMT_CEA_22_720x576I_50HZ	 	0x16
VFMT_CEA_23_720x288P_50HZ	 	0x17
VFMT_CEA_24_720x288P_50HZ	 	0x18
VFMT_CEA_25_720x576I_50HZ	 	0x19
VFMT_CEA_26_720x576I_50HZ	 	0x1a
VFMT_CEA_27_720x288P_50HZ 		0x1b
VFMT_CEA_28_720x288P_50HZ 		0x1c
VFMT_CEA_29_1440x576P_50HZ 		0x1d
VFMT_CEA_30_1440x576P_50HZ	 	0x1e
VFMT_CEA_31_1920x1080P_50HZ 		0x1f
VFMT_CEA_32_1920x1080P_24HZ 		0x20
VFMT_CEA_33_1920x1080P_25HZ 		0x21
VFMT_CEA_34_1920x1080P_30HZ 		0x22
VFMT_CEA_35_2880x480P_60HZ 		0x23
VFMT_CEA_36_2880x480P_60HZ 		0x24
VFMT_CEA_37_2880x576P_50HZ 		0x25
VFMT_CEA_38_2880x576P_50HZ 		0x26
VFMT_CEA_60_1280x720P_24HZ 		0x3c
VFMT_CEA_61_1280x720P_25HZ 		0x3d
VFMT_CEA_62_1280x720P_30HZ	 	0x3e
VFMT_VESA_64_640X480_60 		0x40
VFMT_VESA_65_640X480_75 		0x41
VFMT_VESA_66_800X600_60 		0x42
VFMT_VESA_67_800X600_72 		0x43
VFMT_VESA_68_800X600_75 		0x44
VFMT_VESA_69_800X600_85 		0x45
VFMT_VESA_70_800X600_120_RB 		0x46
VFMT_VESA_71_1024X768_60 		0x47
VFMT_VESA_72_1024X768_70 		0x48
VFMT_VESA_73_1024X768_75 		0x49
VFMT_VESA_74_1024X768_85 		0x4a
VFMT_VESA_75_1024X768_120_RB 		0x4b
VFMT_VESA_76_1152X864_60 		0x4c
VFMT_VESA_77_1152X864_75 		0x4d
VFMT_VESA_78_1280X600_60 		0x4e
VFMT_VESA_79_1280X720_60_DMT 		0x4f
VFMT_VESA_80_1280X720_60_CVT 		0x50
VFMT_VESA_81_1280X720_60_CVT_RB 	0x51
VFMT_VESA_82_1280X720_75_CVT 		0x52
VFMT_VESA_83_1280X720_85_CVT 		0x53
VFMT_VESA_84_1280X768_60 		0x54
VFMT_VESA_85_1280X768_60_RB	 	0x55
VFMT_VESA_86_1280X768_75 		0x56
VFMT_VESA_87_1280X800_60 		0x57
VFMT_VESA_88_1280X800_75 		0x58
VFMT_VESA_89_1280X800_85 		0x59
VFMT_VESA_90_1280X800_120_RB 		0x5a
VFMT_VESA_91_1280X960_60_DMT 		0x5b
VFMT_VESA_92_1280X960_60_CVT 		0x5c
VFMT_VESA_93_1280X960_75_CVT 		0x5d
VFMT_VESA_94_1280X960_85 		0x5e
VFMT_VESA_95_1280X960_120_RB 		0x5f
VFMT_VESA_96_1280X1024_60 		0x60
VFMT_VESA_97_1280X1024_75 		0x61
VFMT_VESA_98_1280X1024_85 		0x62
VFMT_VESA_99_1280X1024_120_RB 		0x63
VFMT_VESA_100_1360X768_60 		0x64
VFMT_VESA_101_1360X768_120_RB 		0x65
VFMT_VESA_102_1366X768_60 		0x66
VFMT_VESA_103_1400X1050_60 		0x67
VFMT_VESA_104_1400X1050_75 		0x68
VFMT_VESA_105_1400X1050_85 		0x69
VFMT_VESA_106_1400X1050_120_RB 		0x6a
VFMT_VESA_107_1440X900_60_DMT 		0x6b
VFMT_VESA_108_1440X900_75 		0x6c
VFMT_VESA_109_1440X900_85 		0x6d
VFMT_VESA_110_1440X900_120_RB 		0x6e
VFMT_VESA_111_1600X900_60_CVT 		0x6f
VFMT_VESA_112_1600X900_60_DMT_RB 	0x70
VFMT_VESA_113_1600X900_75_CVT 		0x71
VFMT_VESA_114_1600X900_85_CVT 		0x72
VFMT_VESA_115_1600X1200_60 		0x73
VFMT_VESA_116_1600X1200_70 		0x74
VFMT_VESA_117_1600X1200_75 		0x75
VFMT_VESA_118_1600X1200_85 		0x76
VFMT_VESA_119_1600X1200_120_RB 		0x77
VFMT_VESA_120_1680X1050_60 		0x78
VFMT_VESA_121_1680X1050_60_RB 		0x79
VFMT_VESA_122_1680X1050_75 		0x7a
VFMT_VESA_123_1680X1050_85 		0x7b
VFMT_VESA_124_1680X1050_120_RB 		0x7c
VFMT_VESA_125_1792X1344_60 		0x7d
VFMT_VESA_126_1792X1344_75 		0x7e
VFMT_VESA_127_1856X1392_60 		0x7f
VFMT_VESA_128_1856X1392_75 		0x80
VFMT_VESA_129_1920X1080_60_DMT	 	0x81
VFMT_VESA_130_1920X1080_60_CVT 		0x82
VFMT_VESA_131_1920X1080_60_CVT_RB 	0x83
VFMT_VESA_132_1920X1200_60_CVT	 	0x84
VFMT_VESA_133_1920X1200_60_DMT_RB 	0x85
VFMT_VESA_134_1920X1200_75 		0x86
VFMT_VESA_135_1920X1200_85 		0x87
VFMT_VESA_136_1920X1440_60 		0x88
VFMT_VESA_137_1920X1440_75 		0x89
VFMT_VESA_138_1920X1440_85_CVT 		0x8a
VFMT_VESA_139_2048X1536_60_CVT 		0x8b
VFMT_VESA_140_2048X1536_75_CVT 		0x8c
VFMT_VESA_141_2048X1536_85_CVT 		0x8d
VFMT_VESA_142_2560X1600_60 		0x8e
VFMT_INVALID 				0xff
```

tablesturn, 2024-01-08:
Thank you for the information (which might also be helpful for other users), but until now it seems that I can not configure the chip to output 400x1280.

Some information:
`width = 800, height = 1280, mode = 0x1100`
Outputs a part of the image two times vertically (CEA 17 with 720x576 50Hz 27MHz 31.25kHz, and yes, a different resolution specified in width/height), and

`width = 640, height = 480, mode = 0x4000`
Outputs the image 2.5 times (so it keeps scanning the pixels of the next frames) (VESA 64 with 640x480 60Hz 25.175MHz 31.5kHz)

I guess the second byte determines something related to color, since changing it to 0x4002 or 0x4003 gives the image a magenta overlay for me.

I figured out the 800x1280 0x1100 by brute forcing. Since the chip seems to only support landscape formats, I don't think it's worth a try to go through all the combinations.

Maybe it is possible to edit the firmware or pass custom timings, but I don't have too much hope in this adapter. It's sold as an USB 3.0 device, has a blue (!) USB 2.0 connector, its mass storage interface is very annoying and the reset logic upon starting is also quite questionable.

rhgndf, 2024-01-09:
@tablesturn @Bastindo : If those modes work, I don't mind accepting it as a PR to support additional modes. However, supporting interlaced modes needs an additional field in `struct ms912x_mode` to indicate it.

One possible way would be to set the mode, dump the registers and set another mode and compare the difference to see if the video mode timings can be set from there.

As for the USB connector thing, I have a connector that has the full usb 3.0 pins, but none of them are connected. It is a common occurrence. Real USB 3.0 devices do exist though in #7 .

rhgndf, 2024-01-12:
I tried dumping the registers. Found a few more modes in [2ffd851](https://github.com/rhgndf/ms912x/commit/2ffd851fa6b3e913b10adcea7d561c7689ba3bbe).