rhgndf/ms912x #12: Segfaults when connected

RealJohnSmith, 2024-01-07:
When the device is connected, there is a segfault and it crashes a current window session. It seems like it crashes the entire USB stack actually.

uname -a:
`Linux [hostname] 6.6.8-2-MANJARO #1 SMP PREEMPT_DYNAMIC Thu Dec 21 16:21:45 UTC 2023 x86_64 GNU/Linux`

relevant dmesg log:

```
[  344.753179] usb 1-3: new high-speed USB device number 8 using xhci_hcd
[  344.894862] usb 1-3: New USB device found, idVendor=534d, idProduct=6021, bcdDevice= 1.10
[  344.894867] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  344.894868] usb 1-3: Product: usb extscreen
[  344.894870] usb 1-3: Manufacturer: USB Display 
[  344.894871] usb 1-3: SerialNumber: 2019BA7160B0
[  344.896997] hid-generic 0003:534D:6021.0007: hiddev97,hidraw5: USB HID v1.10 Device [USB Display  usb extscreen] on usb-0000:00:14.0-3/input0
[  344.904472] [drm] Initialized ms912x 0.0.1 20220101 for 1-3:1.3 on minor 2
[  344.904610] ms912x 1-3:1.3: [drm] Cannot find any crtc or sizes
[  344.904995] ms912x 1-3:1.3: [drm] Cannot find any crtc or sizes
[  355.119213] fbcon: Deferring console take-over
[  355.119217] ms912x 1-3:1.3: [drm] fb1: ms912xdrmfb frame buffer device
[  356.512443] rfkill: input handler enabled
[  360.766080] usb 1-3: USB disconnect, device number 8
[  363.079760] rfkill: input handler disabled
[  375.159132] usb 1-3: new high-speed USB device number 9 using xhci_hcd
[  375.300936] usb 1-3: New USB device found, idVendor=534d, idProduct=6021, bcdDevice= 1.10
[  375.300940] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  375.300942] usb 1-3: Product: usb extscreen
[  375.300943] usb 1-3: Manufacturer: USB Display 
[  375.300945] usb 1-3: SerialNumber: 2019BA7160B0
[  375.304742] hid-generic 0003:534D:6021.0008: hiddev97,hidraw5: USB HID v1.10 Device [USB Display  usb extscreen] on usb-0000:00:14.0-3/input0
[  375.312173] [drm] Initialized ms912x 0.0.1 20220101 for 1-3:1.3 on minor 2
[  375.312486] ms912x 1-3:1.3: [drm] Cannot find any crtc or sizes
[  375.312882] ms912x 1-3:1.3: [drm] Cannot find any crtc or sizes
[  375.479426] fbcon: Deferring console take-over
[  375.479430] ms912x 1-3:1.3: [drm] fb1: ms912xdrmfb frame buffer device
[  377.162642] rfkill: input handler enabled
[  380.109295] rfkill: input handler disabled
[  380.121180] rfkill: input handler enabled
[  380.141493] gsd-usb-protect[33508]: segfault at 8 ip 0000564ea3522187 sp 00007fff75ef71d0 error 4 in gsd-usb-protection[564ea3520000+4000] likely on CPU 5 (core 1, socket 0)
[  380.141508] Code: 1e 00 00 ff 15 4a 5e 00 00 eb a2 0f 1f 84 00 00 00 00 00 48 8b 45 c8 48 8d 15 bd 2a 00 00 be 10 00 00 00 48 8d 3d 95 1e 00 00 <48> 8b 48 08 31 c0 ff 15 1d 5e 00 00 e9 5b ff ff ff 0f 1f 84 00 00
[  383.600879] rfkill: input handler disabled
[  383.605754] rfkill: input handler enabled
[  383.623089] gsd-usb-protect[33851]: segfault at 8 ip 000055afe109f187 sp 00007ffe1b3a2f00 error 4 in gsd-usb-protection[55afe109d000+4000] likely on CPU 0 (core 0, socket 0)
[  383.623103] Code: 1e 00 00 ff 15 4a 5e 00 00 eb a2 0f 1f 84 00 00 00 00 00 48 8b 45 c8 48 8d 15 bd 2a 00 00 be 10 00 00 00 48 8d 3d 95 1e 00 00 <48> 8b 48 08 31 c0 ff 15 1d 5e 00 00 e9 5b ff ff ff 0f 1f 84 00 00
[  385.318363] usb 1-3: USB disconnect, device number 9
[  386.508700] rfkill: input handler disabled
[  395.401149] usb 1-1: USB disconnect, device number 2
[  395.401155] usb 1-1.1: USB disconnect, device number 4
[  395.401157] usb 1-1.1.1: USB disconnect, device number 6
[  395.804038] usb 1-1.2: USB disconnect, device number 5
[  398.472005] usb 1-1: new high-speed USB device number 10 using xhci_hcd
[  398.613687] usb 1-1: New USB device found, idVendor=534d, idProduct=6021, bcdDevice= 1.10
[  398.613692] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  398.613693] usb 1-1: Product: usb extscreen
[  398.613695] usb 1-1: Manufacturer: USB Display 
[  398.613696] usb 1-1: SerialNumber: 2019BA7160B0
[  398.615401] hid-generic 0003:534D:6021.0009: hiddev96,hidraw0: USB HID v1.10 Device [USB Display  usb extscreen] on usb-0000:00:14.0-1/input0
[  398.622331] [drm] Initialized ms912x 0.0.1 20220101 for 1-1:1.3 on minor 2
[  398.622466] ms912x 1-1:1.3: [drm] Cannot find any crtc or sizes
[  398.622794] ms912x 1-1:1.3: [drm] Cannot find any crtc or sizes
[  408.664129] fbcon: Deferring console take-over
[  408.664132] ms912x 1-1:1.3: [drm] fb1: ms912xdrmfb frame buffer device
[  410.178874] rfkill: input handler enabled
[  413.227431] gsd-usb-protect[35884]: segfault at 8 ip 0000558bb43c8187 sp 00007ffd3de6b140 error 4 in gsd-usb-protection[558bb43c6000+4000] likely on CPU 5 (core 1, socket 0)
[  413.227446] Code: 1e 00 00 ff 15 4a 5e 00 00 eb a2 0f 1f 84 00 00 00 00 00 48 8b 45 c8 48 8d 15 bd 2a 00 00 be 10 00 00 00 48 8d 3d 95 1e 00 00 <48> 8b 48 08 31 c0 ff 15 1d 5e 00 00 e9 5b ff ff ff 0f 1f 84 00 00
[  414.673416] usb 1-1: USB disconnect, device number 10

```

`journalctl -k | grep ms912x`:
```
... kernel: [drm] Initialized ms912x 0.0.1 20220101 for 1-3:1.3 on minor 2
... kernel: ms912x 1-3:1.3: [drm] Cannot find any crtc or sizes
... kernel: ms912x 1-3:1.3: [drm] Cannot find any crtc or sizes
```

Can you please help me debug what the issue might be, or fix it?

My setup:
* I am running on a lenovo laptop with nvidia graphics card with one native HDMI output port. 
* The behaviour doesn't seem to be affected by a different HDMI connected in the HDMI port or not. When I plug in only the USB converter (without HDMI plugged into it), the crash doesn't happen, only after I connect HDMI cable. 
* There is no difference between first connecting HDMI and then USB or the other way around. 
* There is no difference if I use USB 2 or 3 port. 
* My system keeps crashing in the loop (as it is trying to reinitialize the USB stack) while the device is connected. After I disconnect it, it recovers and I am prompted with login screen (the previous session had crashed).
* I am using X11 session on my computer, if that makes any difference
* I am running kernel 6.6 and compiled this module at commit 2ffd851fa6b3e913b10adcea7d561c7689ba3bbe (latest on `main` at the time of writing this issue)

Please let me know, if you need additional information.

RealJohnSmith, 2024-01-07:
I downgraded my kernel to 61 and tested with your `kernel-6.1` branch. The issues remains the same, but there is no segfault line in dmesg log. The message `Cannot find any crtc or sizes` is still in both journalctl log and dmesg log and it still doesn't work

rhgndf, 2024-01-07:
The segfault issue might be due to gsd-usb-protect, not the driver. Try filing the issue there.

As for `Cannot find any crtc or sizes`, it means the connector is detected as there is something connected, but there is no EDID information available. If you can display images on the monitor at the correct resolution then there is nothing to worry about since the EDID information may be successfully obtained when the connector is connected better. If not, I have observed this in a dual VGA and HDMI device, where the message is only printed when the VGA port is connected.

RealJohnSmith, 2024-01-08:
No, I can't display anything. I am using HDMI port on the USB dongle (the only one there is, there is no VGA) and I can't display anything on the display. And as I said before, it completely crashes the session (both in kernel 6.6 and 6.1), meaning, my laptop screen and other HDMI-connected screen both go black and laptop boot splash screen is displayed (which I assume is some kind of stale buffer, because when I disconnect the device, the login screen shows up faster that what I would expect if the computer was actually rebooting all the time)

rhgndf, 2024-01-08:
Ok, I have reproduced it with X11 and logging in right after connecting the dongle. This seems like an X11 issue more than anything, since it works in Wayland.

rhgndf, 2024-01-08:
Well, I realised the issue is with the device momentarily unplugging which crashes X11, which may not be what you are facing. Mind trying to use systemd-coredump to see where the crash happens at?

Try executing `systemd-coredump` to see which process crashed, and do `systemd-coredump debug $PID` with the pid that crashes to see the call stack.

RealJohnSmith, 2024-01-08:
> Ok, I have reproduced it with X11 and logging in right after connecting the dongle.

My workflow is a little different - I already have a session and then I plugin the device and then it crashes. I didn't try to plug it in before login. I can try it as well.

RealJohnSmith, 2024-01-08:
Hm, idk if this is related or not, but currently it seems like the USB enumeration is stuck - I disconnected the dongle but lsusb still lists it. Connecting and reconnecting doesn't seem to have any effect.

RealJohnSmith, 2024-01-08:
Confirmed, happens even before login

rhgndf, 2024-01-08:
Seems like there are no debugging symbols in Xorg, so I don't think we can figure out whats wrong through it. It might be useful to try with debuginfod to see get the symbols. 

I only had crashes on unplug, so I'm not entirely sure what is going on.

RealJohnSmith, 2024-01-08:
Can you advise what should I do with the debuginfod /how to use it? I don't have it installed even.

Or what would be the next logical step in debugging/isolating the problem/finding a solution or a workaround?

rhgndf, 2024-01-08:
I'm actually not very sure how to do this, since I have not used it in depth, but here might be some links for different distros:

https://wiki.debian.org/Debuginfod
https://wiki.archlinux.org/title/Debuginfod

The idea would be to find out exactly which line in which source file does it crash at, which hopefully debug symbols should be able to find it out. Then the next step would be to fix it. Since you might be using gnome, it might be helpful to post this issue to the gnome bug tracker.

RealJohnSmith, 2024-01-08:
I am indeed using gnome. So, do you think there is not much we can do about it on the driver's side? Perpahs that it might be using some API in the wrong way? If the bug is in fact, in gnome, there is not much of a chance for a timely fix, I am afraid...

rhgndf, 2024-01-08:
Without knowing why Xorg crashes, I don't think I am able to pinpoint the issue unfortunately. It might be the driver's case, or gnome has a bug, but I am unable to debug this without more information.

rhgndf, 2024-01-09:
I have figured out how to do the debuginfod with coredumpctl, have only used it with gdb briefly before. This command works:

`DEBUGINFO_URLS="..." sudo -E coredumpctl debug $PID`