rhgndf/ms912x #13: Driver works well, but with only two resolutions

rosivaldo, 2024-01-11:
As you can see below, the driver works, it is detecting the USB>VGA adpter and therefore the monitor, but it only detects two resolutions:
[image]

My monitor has the 1366x768 resolution, and I check the code and it's there. 
`        MS912X_MODE(1366,  768, 60, 0x6600, MS912X_PIXFMT_UYVY),`

Any ideais what might be?

rhgndf, 2024-01-11:
The EDID from your monitor isnt being detected. Your cable might not be connected well. Is this a USB->VGA adapter single port adapter or is there a HDMI port beside it? I have encountered this issue if there are two ports and the monitor is connected to the VGA port.

rosivaldo, 2024-01-11:
It's only an USB>VGA adapter.

I was testing with dmesg and journalctl, and found I think it's something with DRM:
```
# dmesg -k
...
[ 2707.040889] usb 3-3: New USB device found, idVendor=534d, idProduct=6021, bcdDevice= 1.00
[ 2707.040899] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2707.040902] usb 3-3: Product: usb extscreen
[ 2707.040905] usb 3-3: Manufacturer: MACROSILICON
[ 2707.040908] usb 3-3: SerialNumber: 2018BA7160A0
[ 2707.043489] hid-generic 0003:534D:6021.0007: hiddev96,hidraw0: USB HID v1.10 Device [MACROSILICON usb extscreen] on usb-0000:00:14.0-3/input0
[ 2707.047100] [drm] Initialized ms912x 0.0.1 20220101 for 3-3:1.3 on minor 6
[ 2707.047285] ms912x 3-3:1.3: [drm] Cannot find any crtc or sizes
[ 2707.047768] ms912x 3-3:1.3: [drm] Cannot find any crtc or sizes
[ 2717.319020] fbcon: Deferring console take-over
[ 2717.319023] ms912x 3-3:1.3: [drm] fb1: ms912xdrmfb frame buffer device
...
```

And journalctl claims about a ms912x_dri.so module is not able to find:
```
# journalctl
...
jan 11 11:21:36 rosivaldo-pc kernel: usb 3-3: new high-speed USB device number 12 using xhci_hcd
jan 11 11:21:36 rosivaldo-pc kernel: usb 3-3: device descriptor read/64, error -71
jan 11 11:21:36 rosivaldo-pc kernel: usb 3-3: New USB device found, idVendor=534d, idProduct=6021, bcdDevice= 1.00
jan 11 11:21:36 rosivaldo-pc kernel: usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
jan 11 11:21:36 rosivaldo-pc kernel: usb 3-3: Product: usb extscreen
jan 11 11:21:36 rosivaldo-pc kernel: usb 3-3: Manufacturer: MACROSILICON
jan 11 11:21:36 rosivaldo-pc kernel: usb 3-3: SerialNumber: 2018BA7160A0
jan 11 11:21:36 rosivaldo-pc kernel: hid-generic 0003:534D:6021.0007: hiddev96,hidraw0: USB HID v1.10 Device [MACROSILICON usb extscreen] on usb-0000:00:14.0-3/input0
jan 11 11:21:36 rosivaldo-pc kernel: [drm] Initialized ms912x 0.0.1 20220101 for 3-3:1.3 on minor 6
jan 11 11:21:36 rosivaldo-pc kernel: ms912x 3-3:1.3: [drm] Cannot find any crtc or sizes
jan 11 11:21:36 rosivaldo-pc kernel: ms912x 3-3:1.3: [drm] Cannot find any crtc or sizes
jan 11 11:21:36 rosivaldo-pc mtp-probe[8973]: checking bus 3, device 12: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-3"
jan 11 11:21:36 rosivaldo-pc mtp-probe[8973]: bus: 3, device: 12 was not an MTP device
jan 11 11:21:36 rosivaldo-pc gnome-shell[2998]: MESA-LOADER: failed to open ms912x: /usr/lib64/dri/ms912x_dri.so: não é possível abrir arquivo compartilhado: Arquivo ou diretório inexistente (search paths /usr/lib64/dri, suffix _dri)
jan 11 11:21:36 rosivaldo-pc mtp-probe[9016]: checking bus 3, device 12: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-3"
jan 11 11:21:36 rosivaldo-pc mtp-probe[9016]: bus: 3, device: 12 was not an MTP device
jan 11 11:21:36 rosivaldo-pc gnome-shell[2998]: Added device '/dev/dri/card6' (ms912x) using atomic mode setting.
jan 11 11:21:36 rosivaldo-pc gnome-shell[2998]: g_hash_table_insert_internal: assertion 'hash_table != NULL' failed
jan 11 11:21:36 rosivaldo-pc gnome-shell[2998]: g_hash_table_steal_extended: assertion 'hash_table != NULL' failed
jan 11 11:21:36 rosivaldo-pc gnome-shell[2998]: **MESA-LOADER: failed to open ms912x: /usr/lib64/dri/ms912x_dri.so**: não é possível abrir arquivo compartilhado: Arquivo ou diretório inexistente (search paths /usr/lib64/dri, suffix _dri)
jan 11 11:21:36 rosivaldo-pc gnome-shell[2998]: Failed to initialize accelerated iGPU/dGPU framebuffer sharing: Not hardware accelerated
jan 11 11:21:36 rosivaldo-pc gnome-shell[2998]: Created gbm renderer for '/dev/dri/card6'
jan 11 11:21:47 rosivaldo-pc kernel: fbcon: Deferring console take-over
jan 11 11:21:47 rosivaldo-pc kernel: ms912x 3-3:1.3: [drm] fb1: ms912xdrmfb frame buffer device
...
```

Also, I've tried drm_info, and it only finds thoose two resolutions:
```
# drm_info
...
├───Connectors
│   └───Connector 0
│       ├───Object ID: 31
│       ├───Type: HDMI-A
│       ├───Status: connected
│       ├───Physical size: 410x230 mm
│       ├───Subpixel: unknown
│       ├───Encoders: {0}
│       ├───Modes
│       │   ├───1024x768@60.00 driver nhsync nvsync 
│       │   └───800x600@60.32 driver phsync pvsync 
│       └───Properties
│           ├───"EDID" (immutable): blob = 38
│           ├───"DPMS": enum {On, Standby, Suspend, Off} = On
│           ├───"link-status": enum {Good, Bad} = Good
│           ├───"non-desktop" (immutable): range [0, 1] = 0
│           ├───"TILE" (immutable): blob = 0
│           └───"CRTC_ID" (atomic): object CRTC = 34
...
```

Anyway, thanks a lot for the support and the excellent work on this driver. I'll be trying other resources as well. I home these information helps you in solving this and/or future issues.

rhgndf, 2024-01-11:
Thanks for the info! It is indeed that the card isn't reading the EDID properly. Try checking the cable, swap a cable or check that it is properly connected.

rhgndf, 2024-01-11:
If you have another monitor, you might want to check whether it works with that. I don't have a USB to VGA adapter to try right now.

rosivaldo, 2024-01-11:
I do have another monitor, but it is a new HDMI monitor, and it works well on the HDMI output. It's really something with the USB>VGA adapter I believe.

rhgndf, 2024-01-12:
I just realised there is the monitor name being displayed and drm_info shows that there is an EDID. So please disregard what I was saying. It seems that it is coincidence that not reading the EDID will also put the same resolutions. In this case:

- Is your monitor 1360x768 instead of 1366x768?
- Is it possible to dump the EDID and check the supported resolutions? Maybe there is something wrong with the EDID.

rosivaldo, 2024-01-12:
I've plugged the monitor in the older laptop so we could have some ground base.

> Is your monitor 1360x768 instead of 1366x768?

The older laptop states it's 1360x768, the 1366x768 resolution I got from LG support page (sorry for that). Here's a screenshot of it:
[image]

 > Is it possible to dump the EDID and check the supported resolutions? Maybe there is something wrong with the EDID.

I don't know if I followed the right steps, but I'll leave below both dumps (the non-working and the working one):

First the non-working (with all the commands):
```
#  find /sys -iname edid
/sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0/drm/card0/card0-HDMI-A-1/edid
/sys/devices/pci0000:00/0000:00:02.0/drm/card1/card1-DP-2/edid
/sys/devices/pci0000:00/0000:00:02.0/drm/card1/card1-eDP-1/edid
/sys/devices/pci0000:00/0000:00:02.0/drm/card1/card1-DP-1/edid
/sys/devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.3/drm/card7/card7-HDMI-A-7/edid
# cp /sys/devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.3/drm/card7/card7-HDMI-A-7/edid /tmp/edid
# cd /tmp && xxd edid 
00000000: 00ff ffff ffff ff00 1e6d ad4b 0101 0101  .........m.K....
00000010: 0215 0103 6a29 1778 ea23 35a4 554d 9b25  ....j).x.#5.UM.%
00000020: 1250 54a5 6a00 314f 454f 614f 0101 0101  .PT.j.1OEOaO....
00000030: 0101 0101 0101 6621 50b0 5100 1b30 4070  ......f!P.Q..0@p
00000040: 3600 96e5 1000 001e 0000 0000 0000 0000  6...............
00000050: 0000 0000 0000 0000 0000 0000 00fd 0038  ...............8
00000060: 4b1e 3d09 000a 2020 2020 2020 0000 00fc  K.=...      ....
00000070: 0057 3139 3433 0a20 2020 2020 2020 00f9  .W1943.       ..
```

And here is the EDID from the laptop that's working fine:
```
# xxd edid
00000000: 00ff ffff ffff ff00 1e6d ad4b 0101 0101  .........m.K....
00000010: 0215 0104 a529 1778 e023 35a4 554d 9b25  .....).x.#5.UM.%
00000020: 1250 54a5 6a00 314f 454f 614f 0101 0101  .PT.j.1OEOaO....
00000030: 0101 0101 0101 6621 50b0 5100 1b30 4070  ......f!P.Q..0@p
00000040: 3600 96e5 1000 001e 0000 0000 0000 0000  6...............
00000050: 0000 0000 0000 0000 0000 0000 00fd 0038  ...............8
00000060: 4b1e 3d09 000a 2020 2020 2020 0000 00fc  K.=...      ....
00000070: 0057 3139 3433 0a20 2020 2020 2020 00c7  .W1943.       ..
```

rhgndf, 2024-01-12:
Ok so now the driver just needs a line supporting 1360x768 containing the correct mode number.

rosivaldo, 2024-01-12:
> Yep... That's what I'm looking for right now. I'll post here when I found the correct values.
> 
> Thanks again for the assistance!

I've just find out that inside your project, you already had it. There's a file named resolutions inside a captures/ folder that contains it.

I added the line and now it's working.

Thank you again for the great job and for the assistance!!

rhgndf, 2024-01-12:
Nice! I didn't realise it wasn't added in.

rhgndf, 2024-01-12:
Added more modes in [2ffd851](https://github.com/rhgndf/ms912x/commit/2ffd851fa6b3e913b10adcea7d561c7689ba3bbe).

closed 2024-01-22 by rhgndf