rhgndf/ms912x #14: What is "vaddr" on line 208 of ms912x_transfer.c in 5.10 branch?

thenightterorx, 2024-01-28:
I am trying to see if the 5.10 branch can work, but it only compiles when I set vaddr to an int or something, otherwise "vaddr undeclared" error. Of course, the driver crashes whenever I try to use it (it will connect and show up as /dev/fb1, but writing anything to it causes the system to crash and idk where logs would be). Is it supposed to be declared as something and its a typo? I don't know C

rhgndf, 2024-01-28:
It is indeed a mistake. I have added vaddr in.

thenightterorx, 2024-01-28:
> It is indeed a mistake. I have added vaddr in.

ok now I was able to copy the error into a text file 
`[   54.837939] [drm] Initialized ms912x 0.0.1 20220101 for 5-6:1.3 on minor 1
[   55.031614] ms912x 5-6:1.3: [drm] fb1: ms912xdrmfb frame buffer device
[   55.031645] usbcore: registered new interface driver ms912x
[  141.165095] BUG: unable to handle page fault for address: ffffffffc051f000
[  141.166449] #PF: supervisor read access in kernel mode
[  141.167807] #PF: error_code(0x0000) - not-present page
[  141.169175] PGD 297615067 P4D 297615067 PUD 297617067 PMD 1075fa067 PTE 0
[  141.170568] Oops: 0000 [#1] SMP NOPTI
[  141.171947] CPU: 23 PID: 225 Comm: kworker/23:1 Tainted: G           OE     5.10.207-1-lts510 #1
[  141.173339] Hardware name: To Be Filled By O.E.M. X570 Phantom Gaming-ITX/TB3/X570 Phantom Gaming-ITX/TB3, BIOS P3.70 10/19/2022
[  141.174763] Workqueue: events drm_fb_helper_dirty_work [drm_kms_helper]
[  141.176162] RIP: 0010:memcpy_fromio+0x23/0x50
[  141.177553] Code: cc cc cc cc cc cc cc 0f 1f 44 00 00 48 85 d2 74 28 40 f6 c6 01 75 2f 48 83 fa 01 76 06 40 f6 c6 02 75 1b 48 89 d1 48 c1 e9 02 <f3> a5 f6 c2 02 74 02 66 a5 f6 c2 01 74 01 a4 e9 69 90 8d 00 66 a5
[  141.179018] RSP: 0018:ffffadf3c08c7c20 EFLAGS: 00010212
[  141.180500] RAX: 0000000000000000 RBX: 0000000000000780 RCX: 0000000000000188
[  141.181973] RDX: 0000000000001e00 RSI: ffffffffc051f000 RDI: ffff9ddb482537e0
[  141.183408] RBP: ffffadf3c08c7cd8 R08: 000000000019940c R09: 0000000000000f00
[  141.184864] R10: 000000000024a284 R11: 0000000000000780 R12: ffffadf3c4259d08
[  141.186294] R13: ffff9ddb48252000 R14: 00000000000000a3 R15: ffffffffc051d820
[  141.187757] FS:  0000000000000000(0000) GS:ffff9de24efc0000(0000) knlGS:0000000000000000
[  141.187758] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  141.187759] CR2: ffffffffc051f000 CR3: 0000000104440000 CR4: 0000000000350ee0
[  141.187760] Call Trace:
[  141.187766]  ? __die_body.cold+0x1a/0x1f
[  141.187770]  ? no_context+0x1a1/0x3a0
[  141.187774]  ? exc_page_fault+0x14a/0x150
[  141.187776]  ? asm_exc_page_fault+0x1e/0x30
[  141.187780]  ? memcpy_fromio+0x23/0x50
[  141.187784]  ms912x_fb_send_rect+0x43b/0x580 [ms912x]
[  141.187788]  ms912x_pipe_update+0xeb/0x180 [ms912x]
[  141.187798]  drm_atomic_helper_commit_planes+0xf1/0x250 [drm_kms_helper]
[  141.187808]  drm_atomic_helper_commit_tail+0x41/0x80 [drm_kms_helper]
[  141.187816]  commit_tail+0xd1/0x130 [drm_kms_helper]
[  141.187825]  drm_atomic_helper_commit+0x116/0x140 [drm_kms_helper]
[  141.187835]  drm_atomic_helper_dirtyfb+0x19a/0x270 [drm_kms_helper]
[  141.187844]  drm_fb_helper_dirty_work+0x181/0x1b0 [drm_kms_helper]
[  141.187847]  process_one_work+0x1b2/0x350
[  141.187848]  worker_thread+0x4d/0x3b0
[  141.187850]  ? rescuer_thread+0x3b0/0x3b0
[  141.187852]  kthread+0x115/0x130
[  141.187854]  ? kthread_associate_blkcg+0xb0/0xb0
[  141.187856]  ret_from_fork+0x22/0x30
[  141.187859] Modules linked in: ms912x(OE) xt_conntrack xt_MASQUERADE nf_conntrack_netlink iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 xt_addrtype iptable_filter br_netfilter bridge stp llc overlay evdi(OE) nct6683 vfat fat snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio snd_hda_intel snd_intel_dspcfg soundwire_intel soundwire_generic_allocation edac_mce_amd soundwire_cadence snd_usb_audio snd_hda_codec btusb snd_usbmidi_lib snd_rawmidi joydev mousedev btrtl snd_hda_core snd_seq_device btbcm amdgpu kvm_amd mc snd_hwdep btintel iwlwifi soundwire_bus bluetooth kvm snd_soc_core wmi_bmof usbhid intel_wmi_thunderbolt ecdh_generic ecc irqbypass crc16 snd_compress crct10dif_pclmul vboxnetflt(OE) crc32_pclmul ac97_bus vboxnetadp(OE) ghash_clmulni_intel snd_pcm_dmaengine aesni_intel cfg80211 crypto_simd snd_pcm cryptd snd_timer glue_helper snd rapl soundcore pcspkr sp5100_tco ccp k10temp thunderbolt i2c_piix4 gpu_sched igb rfkill vboxdrv(OE) dca tpm_crb tpm_tis mac_hid
[  141.187904]  pinctrl_amd tpm_tis_core acpi_cpufreq pkcs8_key_parser lzo_rle nouveau mxm_wmi wmi video i2c_algo_bit ttm drm_kms_helper cec syscopyarea sysfillrect sysimgblt fb_sys_fops drm fuse crypto_user dm_mod loop agpgart nfnetlink zram tpm rng_core ip_tables x_tables serio_raw atkbd libps2 xhci_pci xhci_pci_renesas i8042 serio btrfs blake2b_generic libcrc32c crc32c_generic crc32c_intel xor raid6_pq
[  141.187926] CR2: ffffffffc051f000
[  141.187928] ---[ end trace 56eb60887ed84d0d ]---
[  141.187930] RIP: 0010:memcpy_fromio+0x23/0x50
[  141.187931] Code: cc cc cc cc cc cc cc 0f 1f 44 00 00 48 85 d2 74 28 40 f6 c6 01 75 2f 48 83 fa 01 76 06 40 f6 c6 02 75 1b 48 89 d1 48 c1 e9 02 <f3> a5 f6 c2 02 74 02 66 a5 f6 c2 01 74 01 a4 e9 69 90 8d 00 66 a5
[  141.187932] RSP: 0018:ffffadf3c08c7c20 EFLAGS: 00010212
[  141.187933] RAX: 0000000000000000 RBX: 0000000000000780 RCX: 0000000000000188
[  141.187933] RDX: 0000000000001e00 RSI: ffffffffc051f000 RDI: ffff9ddb482537e0
[  141.187934] RBP: ffffadf3c08c7cd8 R08: 000000000019940c R09: 0000000000000f00
[  141.187935] R10: 000000000024a284 R11: 0000000000000780 R12: ffffadf3c4259d08
[  141.187935] R13: ffff9ddb48252000 R14: 00000000000000a3 R15: ffffffffc051d820
[  141.187936] FS:  0000000000000000(0000) GS:ffff9de24efc0000(0000) knlGS:0000000000000000
[  141.187937] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  141.187938] CR2: ffffffffc051f000 CR3: 0000000104440000 CR4: 0000000000350ee0`

This is what the display looks like
[image]