rhgndf/ms912x #18: How to run on Linux 6.1.21 kernel?

Tiger0215, 2024-05-20:
Does it support running on 6.1.21 kernel or 6.6.x kernel?

How to Resolve Errors When Installing Kernel Modules with DKMS?

`pi@raspberrypi:~/ms912x $ sudo dkms install .`
`Creating symlink /var/lib/dkms/ms912x/0.1/source ->/usr/src/ms912x-0.1`
`DKMS: add completed.`
`Error! echo`
`Your kernel headers for kernel 6.1.21-v8+ cannot be found at`
`/lib/modules/6.1.21-v8+/build or /lib/modules/6.1.21-v8+/source.`
`You can use the --kernelsourcedir option to tell DKMS where it's located.
`

rhgndf, 2024-05-21:
You might need to install kernel headers

Tiger0215, 2024-05-26:
> You might need to install kernel headers

OK! I ran it normally but only got the 800x600 resolution mode.
What I need to display is the EDID of 1024x600. How can I add it?

EDID_MODE:2         ;0-NORMAL, 1-REPLACE, 2-APPEND

;RESOLUTION table is valid when EDID_MODE mode is APPEND
;            VIC   Polarity    Total        Active      Freq        Offset      Sync
;                             H     V      H    V     clk    V     H     V     H     V
RESOLUTION:  150,     7,     1344, 635,   1024, 600,  5121, 6000,  160, 23,   40,   5

closed 2024-05-28 by Tiger0215