rhgndf/ms912x #19: How to add 1024x600 resolution mode？

Tiger0215, 2024-05-28:
I ran it normally but only got the 800x600 resolution mode.
What I need to display is the EDID of 1024x600. How can I add it?

My EDID_MODE:2 ;0-NORMAL, 1-REPLACE, 2-APPEND

;RESOLUTION table is valid when EDID_MODE mode is APPEND
; VIC Polarity Total Active Freq Offset Sync
; H V H V clk V H V H V
RESOLUTION: 150, 7, 1344, 635, 1024, 600, 5121, 6000, 160, 23, 40, 5

rhgndf, 2024-05-28:
I don't think the device can support that mode without full register descriptions. 

The modes here: https://github.com/rhgndf/ms912x/blob/main/ms912x_drv.c#L78 are the built-in modes. Doing any other modes will likely need more documentation.

Tiger0215, 2024-05-29:
> I don't think the device can support that mode without full register descriptions.
> 
> The modes here: https://github.com/rhgndf/ms912x/blob/main/ms912x_drv.c#L78 are the built-in modes. Doing any other modes will likely need more documentation.

Thanks for your quick reply.
I'm not sure what full register descriptions are, I just know I'm using the MS9120 USB expansion display.
I'll try to provide more information, please help me see if it helps.
EDID that works properly on the computer:
`
Color characteristics
  Default color space...... Non-sRGB
  Display gamma............ 2.20
  Red chromaticity......... Rx 0.640 - Ry 0.330
  Green chromaticity....... Gx 0.300 - Gy 0.600
  Blue chromaticity........ Bx 0.150 - By 0.060
  White point (default).... Wx 0.313 - Wy 0.329
  Additional descriptors... None

Timing characteristics
  Horizontal scan range.... 31-32kHz
  Vertical scan range...... 59-60Hz
  Video bandwidth.......... 40MHz
  CVT standard............. Not supported
  GTF standard............. Not supported
  Additional descriptors... None
  Preferred timing......... Yes
  Native/preferred timing.. 1024x600p at 60Hz 
    Modeline............... "1024x600" 51.210 1024 1184 1224 1344 600 612 617 635 +hsync +vsync

Report information
  Date generated........... 2024/5/29
  Software revision........ 2.91.0.1043
  Data source.............. Registry-Active
  Operating system......... 10.0.19045.2

Raw data
  00,FF,FF,FF,FF,FF,FF,00,21,57,D0,FB,81,B1,34,01,16,21,01,03,80,0A,09,78,22,EE,91,A3,54,4C,99,26,
  0F,50,54,01,00,00,01,01,01,01,01,01,01,01,01,01,01,01,01,01,01,01,01,14,00,40,41,58,23,20,A0,28,
  C5,00,00,A4,0B,36,D6,1E,00,00,00,FC,00,30,37,30,2D,55,53,42,2D,4C,43,44,0A,20,00,00,00,FD,00,3B,
  3C,1F,20,04,00,0A,20,20,20,20,20,20,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,C3
`

With your Linux driver, it can display in 800x600 resolution mode, but it's not fully displayed.
XWAYLAND2 connected 800x600+800+0 (normal left inverted right x axis y axis) 100mm x 90mm
   800x600       60.25*+

So I thought maybe there is a universal 1024x600 resolution mode that allows it to display fully.

rhgndf, 2024-05-29:
By full register descriptions it means that we will need the operating details of the device from the manufacturer macrosilicon.

Since we don't have that, it is unlikely the resolution will be supported.

closed 2024-06-06 by Tiger0215