rhgndf/ms912x #21: Is this working with "Ubuntu 22.04.4 LTS"?

protektwar, 2024-10-04:
Hello I'm trying to compile this driver under Ubuntu 22.04.4 LTS. 

I'm using kernel 5.15.0-119-lowlatency and while compiling I get a lot of errors:

` 
make all -j
make CHECK="/usr/bin/sparse" -C /lib/modules/5.15.0-119-lowlatency/build M=/home/user123/projects/otherProjects/ms912x modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-119-lowlatency'
  CC [M]  /home/user123/projects/otherProjects/ms912x/ms912x_registers.o
  CC [M]  /home/user123/projects/otherProjects/ms912x/ms912x_connector.o
  CC [M]  /home/user123/projects/otherProjects/ms912x/ms912x_transfer.o
  CC [M]  /home/user123/projects/otherProjects/ms912x/ms912x_drv.o
In file included from /home/user123/projects/otherProjects/ms912x/ms912x_connector.c:8:
/home/user123/projects/otherProjects/ms912x/ms912x.h:113:66: warning: ‘struct iosys_map’ declared inside parameter list will not be visible outside of this definition or declaration
  113 | int ms912x_fb_send_rect(struct drm_framebuffer *fb, const struct iosys_map *map,
      |                                                                  ^~~~~~~~~
In file included from /home/user123/projects/otherProjects/ms912x/ms912x_registers.c:4:
/home/user123/projects/otherProjects/ms912x/ms912x.h:113:66: warning: ‘struct iosys_map’ declared inside parameter list will not be visible outside of this definition or declaration
  113 | int ms912x_fb_send_rect(struct drm_framebuffer *fb, const struct iosys_map *map,
      |                                                                  ^~~~~~~~~
In file included from /home/user123/projects/otherProjects/ms912x/ms912x_drv.c:19:
/home/user123/projects/otherProjects/ms912x/ms912x.h:113:66: warning: ‘struct iosys_map’ declared inside parameter list will not be visible outside of this definition or declaration
  113 | int ms912x_fb_send_rect(struct drm_framebuffer *fb, const struct iosys_map *map,
      |                                                                  ^~~~~~~~~
In file included from /home/user123/projects/otherProjects/ms912x/ms912x_transfer.c:8:
/home/user123/projects/otherProjects/ms912x/ms912x.h:113:66: warning: ‘struct iosys_map’ declared inside parameter list will not be visible outside of this definition or declaration
  113 | int ms912x_fb_send_rect(struct drm_framebuffer *fb, const struct iosys_map *map,
      |                                                                  ^~~~~~~~~
/home/user123/projects/otherProjects/ms912x/ms912x_transfer.c:105:46: warning: ‘struct iosys_map’ declared inside parameter list will not be visible outside of this definition or declaration
  105 |                                       struct iosys_map *xrgb_buffer,
      |                                              ^~~~~~~~~
/home/user123/projects/otherProjects/ms912x/ms912x_drv.c: In function ‘ms912x_pipe_update’:
/home/user123/projects/otherProjects/ms912x/ms912x_drv.c:182:53: error: ‘shadow_plane_state’ undeclared (first use in this function); did you mean ‘drm_plane_state’?
  182 |                 if (ms912x_fb_send_rect(state->fb, &shadow_plane_state->data[0],
      |                                                     ^~~~~~~~~~~~~~~~~~
      |                                                     drm_plane_state
/home/user123/projects/otherProjects/ms912x/ms912x_drv.c:182:53: note: each undeclared identifier is reported only once for each function it appears in
/home/user123/projects/otherProjects/ms912x/ms912x_drv.c:186:53: error: ‘struct ms912x_device’ has no member named ‘update_rec’; did you mean ‘update_rect’?
  186 |                                            &ms912x->update_rec  t, &rect);
      |                                                     ^~~~~~~~~~
      |                                                     update_rect
/home/user123/projects/otherProjects/ms912x/ms912x_drv.c:186:64: error: expected ‘)’ before ‘t’
  186 |                                            &ms912x->update_rec  t, &rect);
      |                                                                ^~
      |                                                                )
/home/user123/projects/otherProjects/ms912x/ms912x_drv.c:185:43: note: to match this ‘(’
  185 |                         ms912x_merge_rects(&ms912x->update_rect,
      |                                           ^
/home/user123/projects/otherProjects/ms912x/ms912x_drv.c:185:25: error: too few arguments to function ‘ms912x_merge_rects’
  185 |                         ms912x_merge_rects(&ms912x->update_rect,
      |                         ^~~~~~~~~~~~~~~~~~
/home/user123/projects/otherProjects/ms912x/ms912x_drv.c:160:13: note: declared here
  160 | static void ms912x_merge_rects(struct drm_rect *dest, struct drm_rect *r1,
      |             ^~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:297: /home/user123/projects/otherProjects/ms912x/ms912x_drv.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/user123/projects/otherProjects/ms912x/ms912x_transfer.c: In function ‘ms912x_xrgb_to_yuv422_line’:
/home/user123/projects/otherProjects/ms912x/ms912x_transfer.c:113:9: error: implicit declaration of function ‘iosys_map_memcpy_from’ [-Werror=implicit-function-declaration]
  113 |         iosys_map_memcpy_from(temp_buffer, xrgb_buffer, offset, width * 4);
      |         ^~~~~~~~~~~~~~~~~~~~~
/home/user123/projects/otherProjects/ms912x/ms912x_transfer.c: At top level:
/home/user123/projects/otherProjects/ms912x/ms912x_transfer.c:146:65: warning: ‘struct iosys_map’ declared inside parameter list will not be visible outside of this definition or declaration
  146 | static int ms912x_fb_xrgb8888_to_yuv422(void *dst, const struct iosys_map *src,
      |                                                                 ^~~~~~~~~
/home/user123/projects/otherProjects/ms912x/ms912x_transfer.c: In function ‘ms912x_fb_xrgb8888_to_yuv422’:
/home/user123/projects/otherProjects/ms912x/ms912x_transfer.c:152:26: error: storage size of ‘fb_map’ isn’t known
  152 |         struct iosys_map fb_map;
      |                          ^~~~~~
/home/user123/projects/otherProjects/ms912x/ms912x_transfer.c:172:18: error: implicit declaration of function ‘IOSYS_MAP_INIT_OFFSET’ [-Werror=implicit-function-declaration]
  172 |         fb_map = IOSYS_MAP_INIT_OFFSET(src, y1 * fb->pitches[0]);
[... 33 lines cut ...]
  116 | int drm_gem_shmem_vmap(struct drm_gem_shmem_object *shmem, struct dma_buf_map *map);
      |     ^~~~~~~~~~~~~~~~~~
/home/user123/projects/otherProjects/ms912x/ms912x_transfer.c:223:44: error: passing argument 2 of ‘ms912x_fb_xrgb8888_to_yuv422’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  223 |                                            map, fb, rect);
      |                                            ^~~
      |                                            |
      |                                            const struct iosys_map *
/home/user123/projects/otherProjects/ms912x/ms912x_transfer.c:146:76: note: expected ‘const struct iosys_map *’ but argument is of type ‘const struct iosys_map *’
  146 | static int ms912x_fb_xrgb8888_to_yuv422(void *dst, const struct iosys_map *src,
      |                                                    ~~~~~~~~~~~~~~~~~~~~~~~~^~~
/home/user123/projects/otherProjects/ms912x/ms912x_transfer.c:230:37: error: passing argument 1 of ‘drm_gem_shmem_vunmap’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  230 |         drm_gem_shmem_vunmap(fb->obj[0], vaddr);
      |                              ~~~~~~~^~~
      |                                     |
      |                                     struct drm_gem_object *
In file included from /home/user123/projects/otherProjects/ms912x/ms912x_transfer.c:6:
./include/drm/drm_gem_shmem_helper.h:117:56: note: expected ‘struct drm_gem_shmem_object *’ but argument is of type ‘struct drm_gem_object *’
  117 | void drm_gem_shmem_vunmap(struct drm_gem_shmem_object *shmem, struct dma_buf_map *map);
      |                           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:297: /home/user123/projects/otherProjects/ms912x/ms912x_transfer.o] Error 1
make[1]: *** [Makefile:1911: /home/user123/projects/otherProjects/ms912x] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-119-lowlatency'
make: *** [Makefile:15: modules] Error 2
`

rhgndf, 2024-10-06:
Try the kernel-6.1 branch instead.

protektwar, 2024-10-07:
I already did as it was my first try, there are other errors: 
`
> make all -j
make CHECK="/usr/bin/sparse" -C /lib/modules/5.15.0-122-lowlatency/build M=/home/protektwar/projects/otherProjects/ms912x modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-122-lowlatency'
  CC [M]  /home/protektwar/projects/otherProjects/ms912x/ms912x_transfer.o
  CC [M]  /home/protektwar/projects/otherProjects/ms912x/ms912x_drv.o
In file included from /home/protektwar/projects/otherProjects/ms912x/ms912x_transfer.c:7:
/home/protektwar/projects/otherProjects/ms912x/ms912x.h:113:66: warning: ‘struct iosys_map’ declared inside parameter list will not be visible outside of this definition or declaration
  113 | int ms912x_fb_send_rect(struct drm_framebuffer *fb, const struct iosys_map *map,
      |                                                                  ^~~~~~~~~
/home/protektwar/projects/otherProjects/ms912x/ms912x_transfer.c:104:46: warning: ‘struct iosys_map’ declared inside parameter list will not be visible outside of this definition or declaration
  104 |                                       struct iosys_map *xrgb_buffer,
      |                                              ^~~~~~~~~
/home/protektwar/projects/otherProjects/ms912x/ms912x_transfer.c: In function ‘ms912x_xrgb_to_yuv422_line’:
/home/protektwar/projects/otherProjects/ms912x/ms912x_transfer.c:112:9: error: implicit declaration of function ‘iosys_map_memcpy_from’ [-Werror=implicit-function-declaration]
  112 |         iosys_map_memcpy_from(temp_buffer, xrgb_buffer, offset, width * 4);
      |         ^~~~~~~~~~~~~~~~~~~~~
/home/protektwar/projects/otherProjects/ms912x/ms912x_transfer.c: At top level:
/home/protektwar/projects/otherProjects/ms912x/ms912x_transfer.c:145:65: warning: ‘struct iosys_map’ declared inside parameter list will not be visible outside of this definition or declaration
  145 | static int ms912x_fb_xrgb8888_to_yuv422(void *dst, const struct iosys_map *src,
      |                                                                 ^~~~~~~~~
/home/protektwar/projects/otherProjects/ms912x/ms912x_transfer.c: In function ‘ms912x_fb_xrgb8888_to_yuv422’:
/home/protektwar/projects/otherProjects/ms912x/ms912x_transfer.c:151:26: error: storage size of ‘fb_map’ isn’t known
  151 |         struct iosys_map fb_map;
      |                          ^~~~~~
/home/protektwar/projects/otherProjects/ms912x/ms912x_drv.c:10:10: fatal error: drm/drm_fbdev_generic.h: No such file or directory
   10 | #include <drm/drm_fbdev_generic.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:297: /home/protektwar/projects/otherProjects/ms912x/ms912x_drv.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/protektwar/projects/otherProjects/ms912x/ms912x_transfer.c:171:18: error: implicit declaration of function ‘IOSYS_MAP_INIT_OFFSET’ [-Werror=implicit-function-declaration]
  171 |         fb_map = IOSYS_MAP_INIT_OFFSET(src, y1 * fb->pitches[0]);
      |                  ^~~~~~~~~~~~~~~~~~~~~
/home/protektwar/projects/otherProjects/ms912x/ms912x_transfer.c:174:17: error: implicit declaration of function ‘iosys_map_incr’ [-Werror=implicit-function-declaration]
  174 |                 iosys_map_incr(&fb_map, fb->pitches[0]);
      |                 ^~~~~~~~~~~~~~
/home/protektwar/projects/otherProjects/ms912x/ms912x_transfer.c:151:26: warning: unused variable ‘fb_map’ [-Wunused-variable]
  151 |         struct iosys_map fb_map;
      |                          ^~~~~~
/home/protektwar/projects/otherProjects/ms912x/ms912x_transfer.c: At top level:
/home/protektwar/projects/otherProjects/ms912x/ms912x_transfer.c:183:66: warning: ‘struct iosys_map’ declared inside parameter list will not be visible outside of this definition or declaration
  183 | int ms912x_fb_send_rect(struct drm_framebuffer *fb, const struct iosys_map *map,
      |                                                                  ^~~~~~~~~
/home/protektwar/projects/otherProjects/ms912x/ms912x_transfer.c:183:5: error: conflicting types for ‘ms912x_fb_send_rect’; have ‘int(struct drm_framebuffer *, const struct iosys_map *, struct drm_rect *)’
  183 | int ms912x_fb_send_rect(struct drm_framebuffer *fb, const struct iosys_map *map,
      |     ^~~~~~~~~~~~~~~~~~~
In file included from /home/protektwar/projects/otherProjects/ms912x/ms912x_transfer.c:7:
/home/protektwar/projects/otherProjects/ms912x/ms912x.h:113:5: note: previous declaration of ‘ms912x_fb_send_rect’ with type ‘int(struct drm_framebuffer *, const struct iosys_map *, struct drm_rect *)’
  113 | int ms912x_fb_send_rect(struct drm_framebuffer *fb, const struct iosys_map *map,
      |     ^~~~~~~~~~~~~~~~~~~
/home/protektwar/projects/otherProjects/ms912x/ms912x_transfer.c: In function ‘ms912x_fb_send_rect’:
/home/protektwar/projects/otherProjects/ms912x/ms912x_transfer.c:212:44: error: passing argument 2 of ‘ms912x_fb_xrgb8888_to_yuv422’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  212 |                                            map, fb, rect);
      |                                            ^~~
      |                                            |
      |                                            const struct iosys_map *
/home/protektwar/projects/otherProjects/ms912x/ms912x_transfer.c:145:76: note: expected ‘const struct iosys_map *’ but argument is of type ‘const struct iosys_map *’
  145 | static int ms912x_fb_xrgb8888_to_yuv422(void *dst, const struct iosys_map *src,
      |                                                    ~~~~~~~~~~~~~~~~~~~~~~~~^~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:297: /home/protektwar/projects/otherProjects/ms912x/ms912x_transfer.o] Error 1
make[1]: *** [Makefile:1910: /home/protektwar/projects/otherProjects/ms912x] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-122-lowlatency'
make: *** [Makefile:15: modules] Error 2
`

rhgndf, 2024-10-13:
Unfortunately I'm not too familiar with old kernel versions, it is better if you can update your kernel to 6.2 or above.

tiirwaa, 2025-07-19:
On Ubuntu 22.04, I managed to install this version [https://github.com/tiirwaa/ms912x](https://github.com/tiirwaa/ms912x) using the 6.15 kernel, which is the current one. You could try it and see if it works for you.