rhgndf/ms912x #22: Fails to compile on Linux Kernel 6.11.1 due to removed DRM functions

caesrcd, 2024-10-07:
After updating the Linux kernel on my Arch system from version 6.10.10 to 6.11.1, DKMS attempted to recompile the ms912x module but encountered an error. When I tried to manually recompile it, I observed errors related to the removal of drm_do_get_edid and drm_fbdev_generic_setup functions.

**Diffstat v6.10.10 -> v6.11.1**
[include/drm/drm_edid.h](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/diff/include/drm/drm_edid.h?id=v6.11.1&id2=v6.10.10)
[include/drm/drm_fbdev_generic.h](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/diff/include/drm/drm_fbdev_generic.h?id=v6.11.1&id2=v6.10.10)

These functions were removed starting from Linux kernel version 6.11.1, causing the compilation to fail. I would like to know if there is a solution or patch available to make the module compatible with the newer kernel version.

rhgndf, 2024-10-13:
I have updated this in the latest commit.

caesrcd, 2024-10-14:
I tried to compile with your latest update... without success.

```shell
[ms912x]$ make all -j
make CHECK="/usr/bin/sparse" -C /lib/modules/6.11.3-arch1-1/build M=/home/caesrcd/repo/ms912x modules
make[1]: Entering directory '/usr/lib/modules/6.11.3-arch1-1/build'

  CC [M]  /home/caesrcd/repo/ms912x/ms912x_registers.o
  CC [M]  /home/caesrcd/repo/ms912x/ms912x_connector.o
  CC [M]  /home/caesrcd/repo/ms912x/ms912x_transfer.o
  CC [M]  /home/caesrcd/repo/ms912x/ms912x_drv.o
../../../../../home/caesrcd/repo/ms912x/ms912x_connector.c: In function ‘ms912x_connector_get_modes’:
../../../../../home/caesrcd/repo/ms912x/ms912x_connector.c:30:14: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
   30 |         edid = drm_edid_read_custom(connector, ms912x_read_edid, ms912x);
      |              ^
../../../../../home/caesrcd/repo/ms912x/ms912x_drv.c:145:1: warning: no previous prototype for ‘ms912x_pipe_mode_valid’ [-Wmissing-prototypes]
  145 | ms912x_pipe_mode_valid(struct drm_simple_display_pipe *pipe,
      | ^~~~~~~~~~~~~~~~~~~~~~
../../../../../home/caesrcd/repo/ms912x/ms912x_drv.c:155:5: warning: no previous prototype for ‘ms912x_pipe_check’ [-Wmissing-prototypes]
  155 | int ms912x_pipe_check(struct drm_simple_display_pipe *pipe,
      |     ^~~~~~~~~~~~~~~~~
../../../../../home/caesrcd/repo/ms912x/ms912x_transfer.c: In function ‘ms912x_free_request’:
../../../../../home/caesrcd/repo/ms912x/ms912x_transfer.c:39:9: error: implicit declaration of function ‘vfree’; did you mean ‘kvfree’? [-Wimplicit-function-declaration]
   39 |         vfree(request->transfer_buffer);
      |         ^~~~~
      |         kvfree
../../../../../home/caesrcd/repo/ms912x/ms912x_transfer.c: In function ‘ms912x_init_request’:
../../../../../home/caesrcd/repo/ms912x/ms912x_transfer.c:53:16: error: implicit declaration of function ‘vmalloc_32’; did you mean ‘kvmalloc’? [-Wimplicit-function-declaration]
   53 |         data = vmalloc_32(len);
      |                ^~~~~~~~~~
      |                kvmalloc
../../../../../home/caesrcd/repo/ms912x/ms912x_transfer.c:53:14: error: assignment to ‘void *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
   53 |         data = vmalloc_32(len);
      |              ^
make[3]: *** [scripts/Makefile.build:244: /home/caesrcd/repo/ms912x/ms912x_transfer.o] Error 1
make[2]: *** [/usr/lib/modules/6.11.3-arch1-1/build/Makefile:1926: /home/caesrcd/repo/ms912x] Error 2
make[1]: *** [Makefile:224: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.11.3-arch1-1/build'

make: *** [Makefile:15: modules] Error 2
[ble: exit 2]
[ms912x]$ uname -r
6.11.3-arch1-1
[ms912x]$ git log HEAD -n 4 --oneline
2b34181 (HEAD -> main, origin/main, origin/HEAD) Fix edid update
758acc5 Kernel 6.11 support
738aef1 (origin/kernel-6.10) Add support for USB 3 devices
97bced1 Create LICENSE for possible upstreaming
```

caesrcd, 2024-10-23:
Now compiled on the latest commit.

```bash
[repo]$ git clone https://github.com/rhgndf/ms912x.git
Cloning into 'ms912x'...
remote: Enumerating objects: 143, done.
remote: Counting objects: 100% (71/71), done.
remote: Compressing objects: 100% (25/25), done.
remote: Total 143 (delta 57), reused 53 (delta 46), pack-reused 72 (from 1)
Receiving objects: 100% (143/143), 49.28 KiB | 3.28 MiB/s, done.
Resolving deltas: 100% (86/86), done.
[repo]$ cd ms912x/
[ms912x]$ make all -j $(nproc)
make CHECK="/usr/bin/sparse" -C /lib/modules/6.11.4-arch1-1/build M=/home/caesrcd/repo/ms912x modules
make[1]: Entering directory '/usr/lib/modules/6.11.4-arch1-1/build'

  CC [M]  /home/caesrcd/repo/ms912x/ms912x_registers.o
  CC [M]  /home/caesrcd/repo/ms912x/ms912x_connector.o
  CC [M]  /home/caesrcd/repo/ms912x/ms912x_transfer.o
  CC [M]  /home/caesrcd/repo/ms912x/ms912x_drv.o
../../../../../home/caesrcd/repo/ms912x/ms912x_drv.c:145:1: warning: no previous prototype for ‘ms912x_pipe_mode_valid’ [-Wmissing-prototypes]
  145 | ms912x_pipe_mode_valid(struct drm_simple_display_pipe *pipe,
      | ^~~~~~~~~~~~~~~~~~~~~~
../../../../../home/caesrcd/repo/ms912x/ms912x_drv.c:155:5: warning: no previous prototype for ‘ms912x_pipe_check’ [-Wmissing-prototypes]
  155 | int ms912x_pipe_check(struct drm_simple_display_pipe *pipe,
      |     ^~~~~~~~~~~~~~~~~
  LD [M]  /home/caesrcd/repo/ms912x/ms912x.o
  MODPOST /home/caesrcd/repo/ms912x/Module.symvers
  CC [M]  /home/caesrcd/repo/ms912x/ms912x.mod.o
  LD [M]  /home/caesrcd/repo/ms912x/ms912x.ko
  BTF [M] /home/caesrcd/repo/ms912x/ms912x.ko
make[1]: Leaving directory '/usr/lib/modules/6.11.4-arch1-1/build'
```

However, the built module is not loaded:

```bash
[ms912x]$ sudo insmod ms912x.ko 
insmod: ERROR: could not insert module ms912x.ko: Unknown symbol in module
[ms912x]$ journalctl -e | grep ms912x
out 23 19:43:09 archlinux kernel: ms912x: Unknown symbol drm_fbdev_ttm_setup (err -2)
```

insaner, 2025-05-23:
Hi, just tried compiling and it failed because I am running kernel 6.8. Can you wrap the 6.11 code in a `#if` so it compiles for earlier kernels too? I did a git checkout of 738aef1b659927b21be62a71fbe6f296bea86ae0 and compilation worked fine, and it instantly displayed on the external screen attached to the adapter (very cool! thank you!) About to test the rest of it, but just figured I should let you know.

closed 2025-09-20 by caesrcd