rhgndf/ms912x #25: Device not working, Ubuntu 22.04.1, kernel 6.8.0-47-generic

hvico, 2024-12-05:
lsusb:

Bus 001 Device 021: ID 534d:6021 MacroSilicon VGA Display Adapter

dmesg:

[ 4890.100197] hid-generic 0003:534D:6021.0016: hiddev0,hidraw4: USB HID v1.10 Device [USB Display  usb extscreen] on usb-0000:00:14.0-5/input0
[ 4890.107945] [drm] Initialized ms912x 0.0.1 20220101 for 1-5:1.3 on minor 2
[ 4890.108330] ms912x 1-5:1.3: [drm] Cannot find any crtc or sizes
[ 4890.108887] ms912x 1-5:1.3: [drm] Cannot find any crtc or sizes
[ 4900.504712] ms912x 1-5:1.3: [drm] fb1: ms912xdrmfb frame buffer device

If I run ./insmod.sh I get a screen full of garbage (color lines or snow).

Thanks!

hvico, 2024-12-05:
More data:

When I run ./insmod.sh this info appears on dmesg:

[ 5057.603138] [drm] Initialized ms912x 0.0.1 20220101 for 1-1:1.3 on minor 2
[ 5057.670790] EDID block 0 (tag 0x00) checksum is invalid, remainder is 17
[ 5057.670793] 	[00] BAD  00 ff ff ff ff ff ff 00 54 cc 12 86 12 06 24 20
[ 5057.670794] 	[00] BAD  08 22 01 03 80 23 15 78 ea 56 3a a6 57 4e a8 23
[ 5057.670795] 	[00] BAD  0e 4d 54 21 08 00 d1 c0 01 01 01 01 01 01 01 01
[ 5057.670796] 	[00] BAD  01 01 01 01 01 01 02 3a 80 18 71 38 2d 40 58 2c
[ 5057.670797] 	[00] BAD  45 00 5e cd 10 00 00 1e 00 00 00 ff 00 53 36 4c
[ 5057.670797] 	[00] BAD  2d 32 30 32 34 30 36 31 32 0a 00 00 00 fc 00 53
[ 5057.670798] 	[00] BAD  36 2d 4c 0a 20 20 20 20 20 20 20 20 00 00 00 fd
[ 5057.670799] 	[00] BAD  00 38 4c 1e 51 11 00 0a 20 20 20 20 20 20 01 01

rhgndf, 2024-12-18:
It seems like your monitor is returning bad data.