rhgndf/ms912x #30: Failed to build on Arch Linux 6.14

alfredopalhares, 2025-05-13:
First of all, thank you for this, I am trying use this driver for my external screen Uperfect Delta but i would like to build it has is first.

I am building from commit `b502d3f2d7c513ccf37c3cceea51ca40abde4ca4`:
```
-> % git show HEAD
commit b502d3f2d7c513ccf37c3cceea51ca40abde4ca4 (HEAD -> main, origin/main, origin/HEAD)
```

```
DKMS (dkms-3.1.8) make.log for ms912x/0.1 for kernel 6.14.5-arch1-1 (x86_64)
ter 13 mai 2025 13:52:25 WEST

Building module(s)
# command: make -j12 KERNELRELEASE=6.14.5-arch1-1 all
make CHECK="/usr/bin/sparse" -C /lib/modules/6.14.5-arch1-1/build M=/var/lib/dkms/ms912x/0.1/build modules
make[1]: Entering directory '/usr/lib/modules/6.14.5-arch1-1/build'
make[2]: Entering directory '/var/lib/dkms/ms912x/0.1/build'
  CC [M]  ms912x_registers.o
  CC [M]  ms912x_connector.o
  CC [M]  ms912x_transfer.o
  CC [M]  ms912x_drv.o
ms912x_drv.c:66:10: error: ‘const struct drm_driver’ has no member named ‘date’
   66 |         .date = DRIVER_DATE,
      |          ^~~~
In file included from ms912x_drv.c:21:
ms912x.h:16:21: error: initialization of ‘unsigned int’ from ‘char *’ makes integer from pointer without a cast [-Wint-conversion]
   16 | #define DRIVER_DATE "20220101"
      |                     ^~~~~~~~~~
ms912x_drv.c:66:17: note: in expansion of macro ‘DRIVER_DATE’
   66 |         .date = DRIVER_DATE,
      |                 ^~~~~~~~~~~
ms912x.h:16:21: note: (near initialization for ‘driver.driver_features’)
   16 | #define DRIVER_DATE "20220101"
      |                     ^~~~~~~~~~
ms912x_drv.c:66:17: note: in expansion of macro ‘DRIVER_DATE’
   66 |         .date = DRIVER_DATE,
      |                 ^~~~~~~~~~~
ms912x.h:16:21: error: initializer element is not computable at load time
   16 | #define DRIVER_DATE "20220101"
      |                     ^~~~~~~~~~
ms912x_drv.c:66:17: note: in expansion of macro ‘DRIVER_DATE’
   66 |         .date = DRIVER_DATE,
      |                 ^~~~~~~~~~~
ms912x.h:16:21: note: (near initialization for ‘driver.driver_features’)
   16 | #define DRIVER_DATE "20220101"
      |                     ^~~~~~~~~~
ms912x_drv.c:66:17: note: in expansion of macro ‘DRIVER_DATE’
   66 |         .date = DRIVER_DATE,
      |                 ^~~~~~~~~~~
ms912x_drv.c:145:1: warning: no previous prototype for ‘ms912x_pipe_mode_valid’ [-Wmissing-prototypes]
  145 | ms912x_pipe_mode_valid(struct drm_simple_display_pipe *pipe,
      | ^~~~~~~~~~~~~~~~~~~~~~
ms912x_drv.c:155:5: warning: no previous prototype for ‘ms912x_pipe_check’ [-Wmissing-prototypes]
  155 | int ms912x_pipe_check(struct drm_simple_display_pipe *pipe,
      |     ^~~~~~~~~~~~~~~~~
ms912x_drv.c: In function ‘ms912x_usb_probe’:
ms912x_drv.c:278:9: error: implicit declaration of function ‘drm_fbdev_ttm_setup’ [-Wimplicit-function-declaration]
  278 |         drm_fbdev_ttm_setup(dev, 0);
      |         ^~~~~~~~~~~~~~~~~~~
make[4]: *** [/usr/lib/modules/6.14.5-arch1-1/build/scripts/Makefile.build:207: ms912x_drv.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [/usr/lib/modules/6.14.5-arch1-1/build/Makefile:1997: .] Error 2
make[2]: *** [/usr/lib/modules/6.14.5-arch1-1/build/Makefile:251: __sub-make] Error 2
make[2]: Leaving directory '/var/lib/dkms/ms912x/0.1/build'
make[1]: *** [Makefile:251: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.14.5-arch1-1/build'
make: *** [Makefile:15: modules] Error 2

# exit code: 2
# elapsed time: 00:00:01
----------------------------------------------------------------
````

meocoder31099, 2025-05-26:
I'm having the same issue. 
Did you find a fix for it?

DKoTech, 2025-06-30:
Bro, did you make it successfully? You can see my last closed issue. That make this compile successfully.

DKoTech, 2025-06-30:
> I have reviewed the source code from the Linux kernel.
> 
> Linux kernel 6.14 does not have the `drm_fbdev_ttm_setup` method, it has been replaced by the `drm_client_setup` method from `drm/clients/drm_client_setup.h`. I tried replacing the `drm_fbdev_ttm_setup` method with `drm_client_setup` and removing `.date = DRIVER_DATE`, in the code of the `ms912x_drv.c` file. After that, I was able to compile and install the driver, but unfortunately, the driver does not work. I only get a noisy screen, and my entire system becomes very laggy (the system freezes if I modify any display-related settings).

i saw your this comment and try to do it. i also saw a noisy screen, and repo creator looks like not response any issues long time.

tiirwaa, 2025-07-19:
I managed to successfully compile this repo for Linux 6.15, and it includes the mentioned changes. The device works correctly, showing different resolutions and the sound works fine, although it runs a bit slow. You could try this one and see if it works for you: [](https://github.com/tiirwaa/ms912x)

rhgndf, 2026-09-03:
Fixed in latest commits

closed 2026-09-03 by rhgndf